tuxedocomputers/tuxedo-keyboard #7: Confusion about installation and usage

Maldus512, 2018-10-30:
Hello,
I have a CLEVO P650HP3. I would like to use two of its features that do not work out of the box: the airplane mode function key and the multicolor keyboard backlight.c .

I've been fiddling with your driver for about a year now, trying to make everything work. By installing the dkms module from AUR (I use Arch) I've been able to use the airplane mode key but not the backlight. I believed my laptop model was not yet supported so I kinda forgot about it.

Now however I've tried installing this module from Github and found out it works. To use it however I need to disable the other DKMS (from bitbucket, https://bitbucket.org/tuxedocomputers/clevo-xsm-wmi/overview) which correctly manages the airplane key. With the github module alone on the contrary it does not work (not sure if it's not meant to).

Now, the Readme says something about it, but I'm not sure I understand correctly. Quoting:

    Use either method only. Do not combine installation methods, such as starting with the build step 
    below and proceeding to use the same build artifacts with the DKMS module. Otherwise the module 
    built via dkms will fail to load with an exec_format error on newer kernels due to a mismatched
     version magic.

    This is why the DKMS build step begins with a make clean step.

    For convenience, on platforms where DKMS is in use, skip to the DKMS section directly.

1. What are the two methods?
1. Are you referring to the other DKMS, the one from bitbucket/aur?
1. What "DKMS section" are you referring to, and opposed to what (I only see "The DKMS Route", no other option)?

Also, I might be misunderstanding heavily how every tool is supposed to work. Is the Github version supposed to supersede the Bitbucket one? Are those two different drivers?
I've read https://bitbucket.org/tuxedocomputers/clevo-xsm-wmi/issues/44/the-status-and-the-future-of-tuxedo-clevo, but I still find myself confused. Could you please clarify/help me?

Thank you,
Maldus

Brainiarc7, 2018-10-31:
Hello there, 

To clarify:

You can install this module via make, as shown in the 'Build the module' section:

`make clean && make`

However, doing that will result in an `exec_format `error on load should you proceed with using the built modules in that step with the DKMS step below. For that reason, skip that part completely and proceed to the DKMS route as documented.

If you're using this Github repository, consider the bitbucket repository's code obsolete, and do not install the latter. 

The documentation should be updated to clarify that the only installation route to be used is the DKMS route.

Maldus512, 2018-11-01:
Thank you; my confusion arose from the fact that the `make` command already installs the module, without the need to specify `make install`.

Still, while using this module I am able to control the keyboard backlight but not the airplane mode key. With the Bitbucket version it was the opposite. Should I open another issue about this?

closed 2018-11-01 by Maldus512