tuxedocomputers/tuxedo-keyboard #9: restore from ram / hibernation does not restore keyboard light

pointhi, 2018-12-29:
When I leaf the Laptop for a longer time alone and it goes into suspend to ram or hibernation, the keyboard backlight changes to blue and does not restore when using the laptop again

djallet, 2019-01-23:
I have the same issue, with a clevo N151ZU. Everything else run fine, this is a pleasure to be able to change keyboard backlit on linux :-)

Darael, 2019-02-12:
A workaround, albeit an untested one, *if* you use systemd: place the following script in your systemd suspend hooks directory (for me, `/lib/systemd/system-sleep/`, though I've seen it be in `/usr/lib/systemd/system-sleep/`), and make it executable:
```
#!/bin/sh
case $1 in
  (pre)
    for file in color_center color_left color_right color_extra mode brightness state;
    do
      filepath="/sys/devices/platform/tuxedo_keyboard/$file"
      if [ -e $filepath ]
      then
        case $file in
          (color_*)
            prefix="0x"
            ;;
          (*)
            prefix=""
            ;;
        esac
        echo "$file ${prefix}$(cat $filepath)" >> /tmp/tuxedo_keyboard_conf
      fi
    done
    ;;
  (post)
    if [ -e /tmp/tuxedo_keyboard_conf ];
    then
      awk '{ print $2 >("/sys/devices/platform/tuxedo_keyboard/" $1) }' /tmp/tuxedo_keyboard_conf
      rm /tmp/tuxedo_keyboard_conf
    fi
    ;;
esac
```

This should save the keyboard settings to `/tmp/tuxedo_keyboard_conf` when your computer sleeps, and restore them when it wakes.  It will *not*, as written, persist across shutdown/reboot, but it will work for all sleep states including hibernate, hybrid sleep, and so on.

djallet, 2019-02-15:
It works like a charm on eOS, great job Darael ! Thanks you so much.

Darael, 2019-02-15:
Updated slightly to be usable with pm-utils as well as systemd-suspend

Darael, 2019-02-18:
@tuxedochris is there any reason the driver can't use driver power-management hooks (`->prepare()` and `->complete()` might be good candidates) or, failing that, register PM suspend notifiers?

tuxedochris, 2019-02-18:
Hi Darael,

no, there are no reasons why we did not use it. I create a task internally, for testing the power management in the driver. I believe that by the end of March we will have a result for the test.

Cheers
Chris

cowai, 2019-03-26:
With the workaround there is a brief moment where the default settings is used. A better default I think would be state = off.

Darael, 2019-03-27:
@cowai On my laptop, at least, this brief moment is unavoidable when resuming from hibernate - the keyboard defaults to steady blue until it receives instructions from the OS, so until the module is loaded it'll be like that. There's no change to the driver or the workaround that would make that stop, because it happens "before" either of those is loaded.

If, on the other hand, it's happening for resume-from-suspend as well, changing the default (try using `/etc/modprobe.d/tuxedo_keyboard.conf` before editing the driver source code) should fix that part.

cowai, 2019-03-27:
@cowai the settings in /etc/modprobe.d/tuxedo_keyboard.conf is not used on resume. I am using the workaround. On resume-from-suspend there is like a 2 second window where crazy blue colors are used that is built into the firmware. Is there any way to change those builtin settings? Maybe on Windows?

Darael, 2019-04-17:
@cowai I apologise for the delay - so far as I know, those settings cannot be changed. Changing the settings on Windows doesn't help - while those will persist across a soft reboot, anything that involves actually powering down will reset the keyboard to steady blue until some OS's driver (be it from Windows or from Linux) sets it so something else. Indeed, as far as I've been able to determine the firmware has no persistent writable memory, and will *always* set the lights to steady-state, maximum-brightness, 0x0000FF (full blue) on a cold start, no matter what you do.

leaanthony, 2020-02-06:
> Hi Darael,
> 
> no, there are no reasons why we did not use it. I create a task internally, for testing the power management in the driver. I believe that by the end of March we will have a result for the test.
> 
> Cheers
> Chris

Any news on this?

ashh87, 2020-02-14:
I've put in a pull request which restores the keyboard settings on resume. It also prevents the default blue colours which are shown before the driver can take back control after resume.

tuxedoxt, 2020-04-16:
The fix from ashh87 is merged and is now available in v2.0.2.

closed 2020-04-16 by tuxedoxt

freechelmi, 2020-04-16:
As said here https://github.com/tuxedocomputers/tuxedo-keyboard/issues/21
Please Tag the repo so we can test.

tuxedoxt, 2020-04-17:
> As said here #21
> Please Tag the repo so we can test.

Releases will (at least from now on) be tagged and the latest can be found by checking out the release branch.