tuxedocomputers/tuxedo-keyboard #12: tuxedo-backlight-control widget

webketje, 2019-03-06:
Dear @tuxedochris,

I recently purchased a Clevo laptop and found this repository after I was frustrated that I couldn't toggle off the keyboard backlight. I'm very grateful for your (and other contributors') work! However I found it tiresome to manually modify the `/etc/modprobe.d/tuxedo_keyboard` file and created a bash script & GUI (with python3 & python3-tk) for it (installable with .deb package, using rmmod & modprobe).

I would really appreciate you give it a try/ your feedback, so I'll leave the link here: https://github.com/webketje/tuxedo-backlight-control/. Sorry for polluting the issues here, if you're interested I can be reached at the public email in [my profile](https://github.com/webketje/).

Lednerb, 2019-06-06:
Hey @webketje 

I've just tested the Version 0.4 of tuxedo-backlight-control on Ubuntu 18.04.
I'm using a Tuxedo Infinitybook 15 v4.

It's a nice to have tool for the different modes.

However maybe you can add a functionality for the brightness.
At the moment I have the problem that I cannot fix the brightness to a specific level, it always lights up to maximum...

webketje, 2019-06-17:
@Lednerb what you referred to in issue #16 is precisely why I didn't add a brightness toggle (because it feels like duplicate functionality given that brightness can be set through HEX colors). With [release 0.5](https://github.com/webketje/tuxedo-backlight-control/releases/tag/v0.5) you can define your own colors in `/etc/tuxedo-backlight-control/colors.conf`

Fanelia82, 2019-10-22:
Hi

i have made a gnome extension in order to control backlight with (modified) tuxedo module.

https://extensions.gnome.org/extension/1949/clevo-backlight-keyboard/

not perfect but working.

If someone want to help make it better, you are welcome

startedfr0mtheb0tt0m, 2019-10-23:
@Fanelia82 its not working with Gnome 3.34.1 and TUXEDO Keyboard

andreaferioli, 2019-12-22:
Just bought a CLEVO N151CU (Lafitè IV from PC Specialist),
Installed ubuntu 19.10, found a way to set the tuxedo.keyboard and this awesome CONTROL WIDGET.
Now I can control the brightness (with keys Fn+ Fn-), light on\off (Fn*), and switch through 9 basic colours (Fn/).
And with the widget i can surf through different modes.

The only thing I'm not seem able to do is to set the multiple color mode. Is there a way to fix it? 

Thank you for your work

webketje, 2019-12-23:
@andreaferioli this has to do either with the `tuxedo-keyboard` driver not supporting multi-color or, - as I've since updated [in the readme](https://github.com/webketje/tuxedo-backlight-control#custom-colors) - , lack of support for your specific Clevo model (mine is N150ZU)

StevenSeifried, 2020-01-07:
I think this issue can be closed

closed 2020-01-07 by webketje