tuxedocomputers/tuxedo-keyboard #25: Not working.

enescaglayan, 2019-07-30:
i did everything explained in readme.md its not actually working for me.
i am using a clevo computer

heres my terminal log
`enes@enes-ABRA-A5-V12-1:~/Belgeler$ sudo apt-get install git
[sudo] password for enes: 
Paket listeleri okunuyor... Bitti
Bağımlılık ağacı oluşturuluyor       
Durum bilgisi okunuyor... Bitti      
Aşağıdaki paketler otomatik olarak kurulmuş ve artık bu paketlere gerek duyulmuyor:
  dkms libatomic1:i386 libbsd0:i386 libdrm-amdgpu1:i386 libdrm-intel1:i386
  libdrm-nouveau2:i386 libdrm-radeon1:i386 libdrm2:i386 libedit2:i386
  libelf1:i386 libexpat1:i386 libffi6:i386 libgl1:i386 libgl1-mesa-dri:i386
  libglapi-mesa:i386 libglvnd0:i386 libglx-mesa0:i386 libglx0:i386
  libllvm8:i386 libnvidia-common-418 libnvidia-compute-418:i386
  libnvidia-decode-418:i386 libnvidia-encode-418:i386 libnvidia-fbc1-418:i386
  libnvidia-gl-418:i386 libnvidia-ifr1-418:i386 libpciaccess0:i386
  libsensors5:i386 libstdc++6:i386 libvdpau1 libx11-6:i386 libx11-xcb1:i386
  libxau6:i386 libxcb-dri2-0:i386 libxcb-dri3-0:i386 libxcb-glx0:i386
  libxcb-present0:i386 libxcb-sync1:i386 libxcb1:i386 libxdamage1:i386
  libxdmcp6:i386 libxext6:i386 libxfixes3:i386 libxnvctrl0 libxshmfence1:i386
  libxxf86vm1:i386 mesa-vdpau-drivers nvidia-kernel-common-418 nvidia-prime
  nvidia-settings pkg-config screen-resolution-extra vdpau-driver-all
Bu paketleri kaldırmak için 'sudo apt autoremove' komutunu kullanın.
Aşağıdaki ek paketler kurulacak:
  git-man liberror-perl
Önerilen paketler:
  git-daemon-run | git-daemon-sysvinit git-doc git-el git-email git-gui gitk
  gitweb git-cvs git-mediawiki git-svn
Aşağıdaki YENİ paketler kurulacak:
  git git-man liberror-perl
0 paket yükseltilecek, 3 yeni paket kurulacak, 0 paket kaldırılacak ve 4 paket yükseltilmeyecek.
4.969 kB arşiv dosyası indirilecek.
Bu işlem tamamlandıktan sonra 33,9 MB ek disk alanı kullanılacak.
N: '/etc/apt/sources.list.d/' dizinindeki 'linrunner-ubuntu-tlp-disco.list.save' dosyası geçersiz bir dosya uzantısı olduğu için yok sayılıyor
Devam etmek istiyor musunuz? [E/h] E
İndir: 1 http://tr.archive.ubuntu.com/ubuntu disco/main amd64 liberror-perl all 0.17027-2 [26,6 kB]
İndir: 2 http://tr.archive.ubuntu.com/ubuntu disco/main amd64 git-man all 1:2.20.1-2ubuntu1 [835 kB]
İndir: 3 http://tr.archive.ubuntu.com/ubuntu disco/main amd64 git amd64 1:2.20.1-2ubuntu1 [4.107 kB]
22 sn.'de 4.969 kB alındı (231 kB/s)                                           
N: '/etc/apt/sources.list.d/' dizinindeki 'linrunner-ubuntu-tlp-disco.list.save' dosyası geçersiz bir dosya uzantısı olduğu için yok sayılıyor
Daha önce seçili olmayan liberror-perl paketi seçiliyor.
(Veritabanı okunuyor ... 195101 dosya veya dizin kurulu durumda.)
Paket açılacak: .../liberror-perl_0.17027-2_all.deb ...
Paket açılıyor: liberror-perl (0.17027-2) ...
Daha önce seçili olmayan git-man paketi seçiliyor.
Paket açılacak: .../git-man_1%3a2.20.1-2ubuntu1_all.deb ...
Paket açılıyor: git-man (1:2.20.1-2ubuntu1) ...
Daha önce seçili olmayan git paketi seçiliyor.
Paket açılacak: .../git_1%3a2.20.1-2ubuntu1_amd64.deb ...
Paket açılıyor: git (1:2.20.1-2ubuntu1) ...
Ayarlanıyor: liberror-perl (0.17027-2) ...
Ayarlanıyor: git-man (1:2.20.1-2ubuntu1) ...
Ayarlanıyor: git (1:2.20.1-2ubuntu1) ...
Tetikleyiciler işleniyor: man-db (2.8.5-2) ...
enes@enes-ABRA-A5-V12-1:~/Belgeler$ sudo apt-get install dkms
Paket listeleri okunuyor... Bitti
Bağımlılık ağacı oluşturuluyor       
Durum bilgisi okunuyor... Bitti      
dkms zaten en yeni sürümde (2.6.1-4ubuntu2.2).
dkms elle kurulmuş olarak ayarlandı.
Aşağıdaki paketler otomatik olarak kurulmuş ve artık bu paketlere gerek duyulmuyor:
  libatomic1:i386 libbsd0:i386 libdrm-amdgpu1:i386 libdrm-intel1:i386
  libdrm-nouveau2:i386 libdrm-radeon1:i386 libdrm2:i386 libedit2:i386
  libelf1:i386 libexpat1:i386 libffi6:i386 libgl1:i386 libgl1-mesa-dri:i386
  libglapi-mesa:i386 libglvnd0:i386 libglx-mesa0:i386 libglx0:i386
  libllvm8:i386 libnvidia-common-418 libnvidia-compute-418:i386
  libnvidia-decode-418:i386 libnvidia-encode-418:i386 libnvidia-fbc1-418:i386
  libnvidia-gl-418:i386 libnvidia-ifr1-418:i386 libpciaccess0:i386
  libsensors5:i386 libstdc++6:i386 libvdpau1 libx11-6:i386 libx11-xcb1:i386
[... 67 lines cut ...]
Signing module:
 - /var/lib/dkms/tuxedo_keyboard/1/5.0.0-21-generic/x86_64/module/tuxedo_keyboard.ko
Secure Boot not enabled on this system.
cleaning build area...(bad exit status: 2)

DKMS: build completed.
enes@enes-ABRA-A5-V12-1:~/Belgeler/tuxedo-keyboard$ sudo dkms install -m tuxedo_keyboard -v 1

tuxedo_keyboard.ko:
Running module version sanity check.
modinfo: ERROR: missing module or filename.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.0.0-21-generic/updates/dkms/

depmod...

DKMS: install completed.
enes@enes-ABRA-A5-V12-1:~/Belgeler/tuxedo-keyboard$ modprobe tuxedo_keyboard
modprobe: ERROR: could not insert 'tuxedo_keyboard': Operation not permitted
enes@enes-ABRA-A5-V12-1:~/Belgeler/tuxedo-keyboard$ sudo modprobe tuxedo_keyboard
modprobe: ERROR: could not insert 'tuxedo_keyboard': No such device
enes@enes-ABRA-A5-V12-1:~/Belgeler/tuxedo-keyboard$ 
`

brokenax3, 2019-07-30:
Have you tried adding
`tuxedo_keyboard`
to /etc/modules-load.d

enescaglayan, 2019-07-31:
theres even no folder named `tuxedo_keyboard` and if you mean adding it to here it does not exist.

quote from my terminal log
> modprobe: ERROR: could not insert 'tuxedo_keyboard': No such device

brokenax3, 2019-08-24:
Sorry I meant adding the line to the conf file in /etc/modules-load.d

enescaglayan, 2019-09-08:
> Sorry I meant adding the line to the conf file in /etc/modules-load.d

I will try that later thanks for reply :)

closed 2019-09-08 by enescaglayan