tuxedocomputers/tuxedo-keyboard #26: /sys/devices/platform/tuxedo_keyboard//color_* not writable

simply-dennis, 2019-08-09:
I tried to change color and it didn't worked for me.

```
root@dennis-laptop:/sys/devices/platform/tuxedo_keyboard# echo 00ff00 > color_right
bash: echo: write error: Invalid argument
```

```
dennis@dennis-laptop:~/tmp/serviceschicht-eva$ sudo sh -c "echo ff0000 > /sys/devices/platform/tuxedo_keyboard/color_right"
[sudo] password for dennis: 
sh: echo: I/O error
```

Module is loaded succesfully:
```
[ 8955.936671] tuxedo_keyboard: Model 'TUXEDO' found
[ 8955.948429] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input41
[ 8956.012998] tuxedo_keyboard: set_mode on CUSTOM
[ 8956.061182] tuxedo_keyboard: Set brightness on 255
[ 8956.070922] tuxedo_keyboard: Set keyboard state on: 1
```

changing mode via 
```
sudo sh -c "echo 4 > /sys/devices/platform/tuxedo_keyboard/mode"
``` 
works.

tuxedoxt, 2019-08-09:
Hello, according to the examples the hex value is prefixed with `0x`.

Try: `echo 0x00ff00 > color_right`

simply-dennis, 2019-08-09:
0xffffff is working.
The below one was kind of irritating.

```
root@dennis-laptop:/home/dennis# cat /sys/devices/platform/tuxedo_keyboard/color_left 
ffffff
```
In this i didn't though about the 0x in front

closed 2019-08-09 by simply-dennis