tuxedocomputers/tuxedo-keyboard #27: Also not working

hboetes, 2019-08-09:
The dmesg, to prove the module is loaded and detecting a proper keyboard. But… Nothing happens, I’m still looking at a sea of light while watching a movie.

The keyboard shortcuts work, but that’s clumsy fn-F4 cycles though intensity 0 to 5. But I’d like to script it.

The keyboard type is just light, nothing fancy with colours.

```
[107441.834490] tuxedo_keyboard: Model 'N8xEJEK                         ' found
[107441.840766] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input155
[107441.903415] tuxedo_keyboard: set_mode on CUSTOM
[107441.941351] tuxedo_keyboard: Set brightness on 255
[107441.949039] tuxedo_keyboard: Set keyboard state on: 1
[107537.734397] tuxedo_keyboard: Set brightness on 10
[107541.243560] tuxedo_keyboard: Set brightness on 0
[107553.634510] tuxedo_keyboard: Set brightness on 255
[107576.386897] tuxedo_keyboard: Set keyboard state on: 0
[107582.500536] tuxedo_keyboard: Set keyboard state on: 1
[107679.836240] tuxedo_keyboard: Set brightness on 0
[107687.649874] tuxedo_keyboard: Set brightness on 255
[107710.142650] tuxedo_keyboard: Set keyboard state on: 0
[107720.365442] tuxedo_keyboard: Set keyboard state on: 1
[107807.330202] tuxedo_keyboard: set_mode on CYCLE
[107814.449540] tuxedo_keyboard: set_mode on CUSTOM
[107821.186976] tuxedo_keyboard: set_mode on RANDOM_COLOR

```

tuxedoxt, 2020-10-06:
Hello, the boring answer is that white only keyboards currently aren't supported. Only RGB keyboards are.

The control of the original basic version was (and still is) a little rugged, it tries to control everything as if it is an RGB keyboard, that's why it is reporting it in the log. I'd like to provide an implementation for white only ones also eventually but still missing some pieces, and time for that. I'll make a note of interest but closing the issue.

closed 2020-10-06 by tuxedoxt

hboetes, 2020-10-06:
You could consider leaving the ticket open and prodding me in case you have stuff to test...

TamaMcGlinn, 2021-01-22:
But, how do you know if your keyboard is supported? For me, dmesg tells me this:

```
tuxedo_keyboard: Model 'P64_HJ,HK1              ' found
```

The readme should start by explaining which models are supported and how to check this.