tuxedocomputers/tuxedo-keyboard #30: Change colour keyboard shortcut kills module

ryan-seymour, 2019-08-21:
I've installed the module using the dkms method and everything seems to work well except for the change color keyboard shortcut. When trying to use this, the back-light switches off and seems to kill the module as no other keyboard shortcuts will then respond until I reload the module.
I've written a script to cycle through the default colours and bound this to another keyboard shortcut for now but it'd be nice to get the proper keyboard shortcut working if possible.

My system is a Clevo NH50_70RH according to lshw.

Zeroneca, 2019-08-26:
Can you post your dmesg output?

ryan-seymour, 2019-08-26:
[dmesg.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/3541166/dmesg.log)
The last 4 lines are the output when hitting the change color shortcut.

Zeroneca, 2019-08-26:
I'm not sure, but eventually your laptop uses different wmi codes for your fn keys. Can you try recompiling with [this patch](https://gist.github.com/bb42ca984c32fc8f6e255cb44651dfd2) ?

I don't know if that works, just try

ryan-seymour, 2019-08-26:
That didn't work but it got me looking and I think I've fixed the issue.
In the function `set_next_color_whole_kb`, I noticed the following commented lines:
`//	if (everything_succesfull){
//		kbd_led_state.whole_kbd_color = new_color_id;
//	}`
It looks like the new color code is never being set after the color is changed so I think it was actually working but was repeatedly setting the color to the default code (0) which is black. I just added the line `kbd_led_state.whole_kbd_color = new_color_id;`, recompiled and installed. Now the shortcuts now all seem to be working.

Zeroneca, 2019-08-26:
Are you using an old version of the module? This commented lines aren't present in the current version

ryan-seymour, 2019-08-27:
Yes, I cloned the repo down before that change.
Thanks for your help anyway.

closed 2019-08-27 by ryan-seymour