tuxedocomputers/tuxedo-keyboard #32: Auto-Off (Timed)

autotechmusic, 2019-09-05:
Greetings,

In Windows, my laptop can shut off the LED keyboard light after a set time (5 min - 20 min) with the control center software.

Does this module have an auto-off setting that we can insert into the conf file.  Please adivse.
Thanks.

AT

tuxedoxt, 2020-05-25:
> Greetings,
> 
> In Windows, my laptop can shut off the LED keyboard light after a set time (5 min - 20 min) with the control center software.
> 
> Does this module have an auto-off setting that we can insert into the conf file. Please adivse.
> Thanks.
> 
> AT

Hello, this is currently not supported but the feature request is relatively often asked for and duly noted ;)

autotechmusic, 2020-05-26:
Looking forward to seeing that in action.

Cheers

closed 2020-05-26 by autotechmusic