tuxedocomputers/tuxedo-keyboard #34: Module isn't working

qoheniac, 2019-09-30:
I followed the DKMS route but my keyboard backlight is still not working. Here are some information:

```
# dkms --version
dkms:2.6
# uname -r
5.2.13-200.fc30.x86_64
# journalctl -qkg tuxedo --no-hostname | cut -c 17-
kernel: DMI: TUXEDO Computers TUXEDO InfinityBook 14v1/X35, BIOS 18.13 09/30/2016
kernel: tuxedo_keyboard: loading out-of-tree module taints kernel.
kernel: tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
kernel: tuxedo_keyboard: No known WMI event notification GUID found
kernel: tuxedo_keyboard: No known WMI event notification GUID found
```

qoheniac, 2020-04-29:
I still get the error above under Fedora 32 and with version 2.0.2. What does 

> No known WMI event notification GUID found

mean? `modprobe` ends with the following error message:

`modprobe: ERROR: could not insert 'tuxedo_keyboard': No such device`

tuxedoxt, 2020-04-29:
> I still get the error above under Fedora 32 and with version 2.0.2. What does
> 
> > No known WMI event notification GUID found
> 
> mean? `modprobe` ends with the following error message:
> 
> `modprobe: ERROR: could not insert 'tuxedo_keyboard': No such device`

This means that the module didn't find what it needs to get the key events and control the keyboard backlight. It is not used for all kinds of backlight. At the moment it supports the "multicolor areas" type.

qoheniac, 2020-04-29:
I own an InfinityBook 14v1 that came with Ubuntu and a working white keyboard backlight. After installing Fedora everything worked just fine out of the box, except for the keyboard backlight. Do you know how I get it to work under Fedora?

closed 2023-12-11 by tuxedo-bot