tuxedocomputers/tuxedo-keyboard #40: Makefile's KDIR variable breaks dkms.

huyizheng, 2020-01-14:
The `dkms.conf` file has this line:
```
MAKE[0]="make -C . KERNELDIR=/lib/modules/${kernelver}/build"
```
Which means that during `dkms install` process, the target kernel is send to `make` command by variable `KERNELDIR`.

However, in `Makefile`, there's this line:
```
KDIR := /lib/modules/$(shell uname -r)/build
```
And the rest of `Makefile` uses this `KDIR` variable to build kernel modules. Which means that it'll always build against current running kernel, rather than the target kernel. So the `dkms install` will always fails when upgrading kernel.

To fix this bug, simple delete the line `KDIR := ...` in `Makefile`, and replace all `KDIR` with `KERNELDIR` in the rest of `Makefile`. Or change like this:
```
if [[ -z $KERNELDIR ]]
  KERNELDIR := /lib/modules/$(shell uname -r)/build
fi
```

Zeroneca, 2020-01-14:
The KDIR variable was reintroduced in 94b275f because removing it broke the dkms less install.

tuxedoxt, 2020-04-16:
KDIR should as of now be used consistently in Makefile and dkms.conf, let me know if this is still an issue.

closed 2020-04-16 by tuxedoxt