tuxedocomputers/tuxedo-keyboard #47: Build module on Fedora 31 fails with sed: can't read dkms.conf: No such file or directory

msandres13, 2020-04-18:
on a fresh up to date Fedora 31 I am trying to build this module.
Output of make
make -C /lib/modules/5.5.17-200.fc31.x86_64/build M=/root/tuxedo-keyboard modules
make[1]: Entering directory '/usr/src/kernels/5.5.17-200.fc31.x86_64'
sed: can't read dkms.conf: No such file or directory
sed: can't read dkms.conf: No such file or directory
sed: can't read dkms.conf: No such file or directory
sed: can't read dkms.conf: No such file or directory
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/5.5.17-200.fc31.x86_64'
[root@raptor tuxedo-keyboard]# dnf update
Last metadata expiration check: 2:20:06 ago on Sat 18 Apr 2020 06:14:40 PM CEST.
Dependencies resolved.
Nothing to do.
Complete!
It looks like for some reason dkms.conf cannot be located.
In the makefile I can see the failing commands.
VER := $(shell sed -n 's/^PACKAGE_VERSION=\([^\n]*\)/\1/p' dkms.conf)
MODULE_NAME := $(shell sed -n 's/^PACKAGE_NAME=\([^\n]*\)/\1/p' dkms.conf)

dkms.conf is there in the same directory as the Makefile.

msandres13, 2020-04-18:
Workaround was to put the fully qualified path for dkms.conf in the sed command.

tuxedoxt, 2020-04-20:
Hello,
VER and MODULE_NAME are not used in the normal make/make clean build process. It is only used for dkms install/packaging purposes. So unless it interrupts the building process it can be ignored. (Should better be hidden in this case.) You can look in the src folder and see if you got the .ko as expected, if so all's well.

edent, 2020-05-02:
I also had this problem on Pop!_OS / Ubuntu.

After cloning the project and checking out the branch, you need to edit the `Makefile`.  Find the lines with
```
# Package version and name from dkms.conf
VER := $(shell sed -n 's/^PACKAGE_VERSION=\([^\n]*\)/\1/p' dkms.conf)
MODULE_NAME := $(shell sed -n 's/^PACKAGE_NAME=\([^\n]*\)/\1/p' dkms.conf)
```
Change `dkms.conf` to the full path of your git download. For example
```
# Package version and name from dkms.conf
VER := $(shell sed -n 's/^PACKAGE_VERSION=\([^\n]*\)/\1/p' /home/MYNAME/git/tuxedo-keyboard/dkms.conf)
MODULE_NAME := $(shell sed -n 's/^PACKAGE_NAME=\([^\n]*\)/\1/p' /home/MYNAME/git/tuxedo-keyboard/dkms.conf)
```

Tuxedo is a great utility - thanks for developing it :-)

tuxedoxt, 2020-05-25:
The error output should, as of now, be hidden. If unresolved please reopen.

closed 2020-05-25 by tuxedoxt