tuxedocomputers/tuxedo-keyboard #51: Keep Lighting Always Turned off

duclos-cavalcanti, 2020-08-31:
Hello,

I am running Manjaro KDE on a Manjaro InfinityBook Pro 15.

All the backlighting shortcuts work perfectly and I am very happy with the options available.

However, currently when I turn the computer on or when it wakes up from hibernation, the backlighting of the Keyboard turns on as well. When turning on the computer from zero (booting), there is this effect of switching between colors once at a time, like a "dance".

I would like however to have the backlighting **always off**, completely, even when being booted. Then, when needed, I could press the necessary shortcut to turn it on.

I tried both changing the "state" variable in the "tuxedo_keyboard.conf" to 0. Something like:

"options tuxedo_keyboard **state=0** mode=50 brightness=50 color_left=0xFFFFFF color_center=0xFFFFFF color_right=0xFFFFFF"

This turns off the KB backlighting some seconds **after** turning the computer on. However the dancing effect of colors still takes place when booting, I would like to have that off as well. 

Meaning, the default being: "Always turned off, only turn on when manually asked to", not the other way around.

Thank you

Zeroneca, 2020-08-31:
I don't think that behavior ist possible with the driver. The initial dance like mode is controlled by the firmware before the driver even loads. AFAIK the driver can't overwrite that behavior.

duclos-cavalcanti, 2020-08-31:
Thanks for the quick reply. I understand.

just for the sake of asking, do you know if I could myself change that behavior through the firmware? or is it probably impossible?

Zeroneca, 2020-09-01:
The driver isn't detecting hibernation. You could trigger a script that turns the keyboard lighting off via the sysfs interface after waking up from hibernation. But I guess the keyboard will flash up blue for a short time until your script gets called.

I don't know about any way to change the firmware code. I may look into it the next weeks and leave a comment when I find out more.

closed 2023-12-11 by tuxedo-bot