tuxedocomputers/tuxedo-keyboard #54: Module tuxedo_keyboard not found

duclos-cavalcanti, 2020-09-25:
Hi, 

I own a Manjaro Infinitybook Pro 15 which came shipped with Manjaro KDE. I re-installed a custom Manjaro system using Manjaro Architect and wished to have the same backlighting used on my KDE system before.

I took the non-DKMS route.

git clone ..
cd tuxedo-keyboard
git checkout release

**Output of "make clean && make"**

make -C /lib/modules/5.4.67-1-MANJARO/build M=/home/duclos/Documents/tuxedo-keyboard clean
make[1]: Entering directory '/usr/lib/modules/5.4.67-1-MANJARO/build'
make[1]: Leaving directory '/usr/lib/modules/5.4.67-1-MANJARO/build'
make -C /lib/modules/5.4.67-1-MANJARO/build M=/home/duclos/Documents/tuxedo-keyboard modules
make[1]: Entering directory '/usr/lib/modules/5.4.67-1-MANJARO/build'
  CC [M]  /home/duclos/Documents/tuxedo-keyboard/./src/tuxedo_keyboard.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /home/duclos/Documents/tuxedo-keyboard/./src/tuxedo_keyboard.mod.o
  LD [M]  /home/duclos/Documents/tuxedo-keyboard/./src/tuxedo_keyboard.ko
make[1]: Leaving directory '/usr/lib/modules/5.4.67-1-MANJARO/build'

**Output of modprobe tuxedo_keyboard:**
modprobe: FATAL: Module tuxedo_keyboard not found in directory /lib/modules/5.4.67-1-MANJARO

I understand that the module is not in the correct location, but I do not know what to change to what to move in the system to make it find it.

Any help is appreciated, thank you

duclos-cavalcanti, 2020-09-25:
Figured it out!

closed 2020-09-25 by duclos-cavalcanti

jayvhaile, 2023-11-05:
i am facing the same issue

jayvhaile, 2023-11-05:
how did you solve it?