tuxedocomputers/tuxedo-keyboard #56: Clevo PC50DC

akayueh, 2020-11-26:
I am trying to get this running, also the screen does indicate a keyboard light been press, as well as dmesg showing increase and decrease, nothing happened.

it seem that the key is not recognised?

```
[  803.281729] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
[  803.281735] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
```

tuxedoxt, 2020-11-26:
Hello,

if there is indication that it should change then it seems that the keys are recognized. If it is a TUXEDO device I am pretty sure this model should work with tuxedo-keyboard alone. If it is not a TUXEDO device, some models also support different keyboard backlights. One hint is to check `lsusb` since others tend to be usb connected.

akayueh, 2020-11-27:
ok, I will do more investigation

tuxedoxt, 2020-11-27:
For your device `tuxedo-keyboard` is stand-alone regarding WMI control. We currently have a new separate module/addition for some USB drivers called `tuxedo-keyboard-ite`. If you are lucky the combination of the two works out of the box. If not, you'll need to do more investigating.

akayueh, 2020-11-27:
```
stevenyu@pop-os:~$ sudo lsusb
[sudo] password for stevenyu: 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 5986:9102 Acer, Inc BisonCam,NB Pro
Bus 001 Device 002: ID 048d:8910 Integrated Technology Express, Inc. ITE Device(829x)
Bus 001 Device 005: ID 8087:0026 Intel Corp. 
Bus 001 Device 004: ID 06cb:00c7 Synaptics, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
stevenyu@pop-os:~$ cat /proc/bus/input/devices
I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
U: Uniq=
H: Handlers=kbd event0 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0003 Version=0000
N: Name="Sleep Button"
P: Phys=PNP0C0E/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0E:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=4000 0 0

I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input2
U: Uniq=
H: Handlers=event2 
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input3
U: Uniq=
H: Handlers=kbd event3 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input4
U: Uniq=
H: Handlers=sysrq kbd event4 leds 
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
[... 179 lines cut ...]

I: Bus=0018 Vendor=06cb Product=cd79 Version=0100
N: Name="SYNA1202:00 06CB:CD79 Mouse"
P: Phys=i2c-SYNA1202:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-SYNA1202:00/0018:06CB:CD79.0001/input/input34
U: Uniq=
H: Handlers=mouse0 event23 
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
B: REL=3
B: MSC=10

I: Bus=0018 Vendor=06cb Product=cd79 Version=0100
N: Name="SYNA1202:00 06CB:CD79 Touchpad"
P: Phys=i2c-SYNA1202:00
S: Sysfs=/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-SYNA1202:00/0018:06CB:CD79.0001/input/input35
U: Uniq=
H: Handlers=mouse1 event24 
B: PROP=1
B: EV=1b
B: KEY=e520 30000 0 0 0 0
B: ABS=2e0800000000003
B: MSC=20
```

tuxedoxt, 2020-11-27:
> `Bus 001 Device 002: ID 048d:8910 Integrated Technology Express, Inc. ITE Device(829x)`

Yes, you have the USB device version. Fortunately we have implemented a basic driver for this in the other module :) Try `tuxedo-keyboard-ite` in combination with `tuxedo-keyboard`.

akayueh, 2020-11-27:
ok, I will check it out.

There is another module which seems to be an older version from you using the ACPI, e.g. https://github.com/KarthikJay/clevo-keyboard, I  think I tried and I can see the module been loaded in the kernel, but it also doesn't work.

akayueh, 2020-11-27:
this one?

https://deb.tuxedocomputers.com/ubuntu/pool/main/t/tuxedo-keyboard-ite/

akayueh, 2020-11-27:
> > `Bus 001 Device 002: ID 048d:8910 Integrated Technology Express, Inc. ITE Device(829x)`
> 
> Yes, you have the USB device version. Fortunately we have implemented a basic driver for this in the other module :) Try `tuxedo-keyboard-ite` in combination with `tuxedo-keyboard`.

Wow, it works...

is there a way to combine the two as one single package?

tuxedoxt, 2020-11-27:
> ok, I will check it out.
> 
> There is another module which seems to be an older version from you using the ACPI, e.g. https://bitbucket.org/tuxedocomputers/clevo-xsm-wmi/downloads/, I think I tried and I can see the module been loaded in the kernel, but it also doesn't work.

Ah wow this looks severely outdated, if you installed this please uninstall or it might interfere with the newer ones.

> Wow, it works...

> is there a way to combine the two as one single package?

Great! Yes, it might happen soon. For now it is what it is, enjoy:)

closed 2020-11-27 by tuxedoxt