tuxedocomputers/tuxedo-keyboard #57: Q: where does tuxedo-keyboard store logs? (like TUXEDO_ERROR)

webketje, 2020-12-10:
For an issue on tuxedo-backlight-control, [a user seems to have an incomplete SysFs installation](https://github.com/webketje/tuxedo-backlight-control/issues/10).
Looking at the code of this repo there should be a trace of issues with the message `Sysfs attribute file creation failed for state`

tuxedoxt, 2020-12-11:
Hello, the messages info/err/debug etc are normal kernel log messages see: https://www.kernel.org/doc/html/latest/core-api/printk-basics.html. So depending on what on the system is storing these is where they can be found. `journalctl` does it as far as I know, depending on system it also saves previous boots.

webketje, 2020-12-11:
Ok thanks closing this issue

commit dd774d8b (2022-04-25): Merge branch '57-stellaris-intel-gen-4-support' into 'master'

Resolve "Stellaris Intel gen 4 support"

Closes #57

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!26
closed 2020-12-11 by webketje