tuxedocomputers/tuxedo-keyboard #59: Broken Support for TongFang GMxMPxW

fedevx, 2020-12-19:
Hello!
I have a TongFang laptop (GM5MP0W), which is experiencing issues with the tuxedo_io module.

With the module loaded, the keyboard's FN combos work with the exception of the backlight. Pressing the keys is recognized by GNOME as the correct combination to increase or decrease brightness as the symbol is displayed on-screen. The keyboard backlight brightness, however, doesn't change. On the journal, I see the following entry coming-up whenever I press the combo:

`gsd-power[3073]: gsd_power_backlight_abs_to_percentage: assertion 'max > min' failed`

Commenting out the following line in tongfang_wmi.h fixes the issue but I wonder if that is breaking something else because for example, while changing the backlight works, the on-screen symbol doesn't show anymore.

`uw_ec_write_addr(0x41, 0x07, 0x01, 0x00, &reg_write_return);`

It also made sense to then comment out

`uw_ec_write_addr(0x41, 0x07, 0x00, 0x00, &reg_write_return);`

Fan control (which seems was the reason behind setting manual mode) is apparently following the profiles. I'm not sure this model requires entering "manual mode" and if it does, then I'd like to try fixing the keyboard backlight issue.

As for the keyboard itself, is an ITE keyboard supporting per-key RGB colors. Here are the details:

```
lsusb -d 048d:6004 -v

Bus 001 Device 003: ID 048d:6004 Integrated Technology Express, Inc. ITE Device(8291)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x048d Integrated Technology Express, Inc.
  idProduct          0x6004 
  bcdDevice            0.03
  iManufacturer           1 ITE Tech. Inc.
  iProduct                2 ITE Device(8291)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      23
          Report Descriptor: (length is 23)
            Item(Global): Usage Page, data= [ 0x89 0xff ] 65417
                            (null)
            Item(Local ): Usage, data= [ 0x10 ] 16
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Report ID, data= [ 0x5a ] 90
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x10 ] 16
            Item(Main  ): Feature, data= [ 0x00 ] 0
                            Data Array Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      31
          Report Descriptor: (length is 31)
            Item(Global): Usage Page, data= [ 0x03 0xff ] 65283
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x40 ] 64
            Item(Local ): Usage, data= [ 0x20 ] 32
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x21 ] 33
                            (null)
            Item(Main  ): Output, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Local ): Usage, data= [ 0x22 ] 34
                            (null)
            Item(Global): Report Count, data= [ 0x08 ] 8
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

Anything I should be testing out which might help to figure out the issue?

Thanks!

tuxedoxt, 2020-12-22:
Hello,

I could argue that it is not broken since it was not yet supported ;)

I would not suggest commenting out the "manual mode setting". It will most likely break fan control functionality.

I see that your keyboard backlight usb device has a new product id. Please try updating to version `0.2.1` of `tuxedo-keyboard-ite`. The driver is still pretty basic and does not yet have interfaces for what the keyboard backlight is capable of. However it should provide basic white backlight and brightness control.

More to come..

fedevx, 2020-12-24:
> Hello,
>
> I could argue that it is not broken since it was not yet supported ;)
>
> I would not suggest commenting out the "manual mode setting". It will most likely break fan control functionality.
>
> I [...]

Thanks for looking at this. I've been using [ite8291r3](https://github.com/pobrn/ite8291r3-ctl) which is a user-space driver. It works perfectly fine to set the key colors and keyboard backlight brightness. What happens is that the moment tuxedo-io loads, the keys to change the backlight brightness stop working.

I'll test tuxedo-keyboard-ite to see if that makes a difference and update this thread back.

fedevx, 2020-12-27:
tuxedo-keyboard-ite as you mentioned, provides white backlight and brightness control only. Is there a way to get the functionality of [ite8291r3](https://github.com/pobrn/ite8291r3-ctl) into the tuxedo-keyboard-ite driver? At the moment ite8291r3 unloads the kernel driver upon execution (as otherwise the resource is locked) so while it allows changing the colors and the backlight brightness (via command line), changing backlight brightness with the Fn shortcut stops working.

If there is, I could help testing this.

tuxedoxt, 2021-01-18:
Correct, `tuxedo-keyboard-ite` currently only supports white backlight. As hinted above..

> The driver is still pretty basic and does not yet have interfaces for what the keyboard backlight is capable of. However it should provide basic white backlight and brightness control.
> 
> More to come..

Features will be extended in the future but for now it is what it is.

I only know the alternative driver is a user-space driver. The hidraw user-space interface is still there although the kernel mode driver is loaded, maybe there is a way they could co-exist?

I will close this for now as the main issue was sorted out and rest is planned features for the future.

closed 2021-01-18 by tuxedoxt