tuxedocomputers/tuxedo-keyboard #60: Issues with Pulse 15

kladnik, 2020-12-22:
Hello,

I am having two small issues with this model:
- When disabling the touchpad the light should be turned on, but it never is
- Sometimes the backlight turn itself on automatically (usually when waking up)

Using Ubuntu 20.10, but had same issues with 20.04.

Thanks!

Darael, 2020-12-22:
In general you can't avoid the light turning on at powerup, which may include from sleep states. That's a firmware issue that the driver can't override because it happens before the driver comes into play. What you *can* do is add a wake-up hook to turn it back off automatically (how you do this will depend partly on what sleep utility you use).

tuxedoxt, 2020-12-22:
>     * When disabling the touchpad the light should be turned on (small indicator that it's off), but it never is

Known and a solution is in the works.

>     * Sometimes the backlight turn itself on automatically (usually when waking up)

Currently there is no extra driver for this device since the built-in firmware functionality is always on and does the logic for the keyboard backlight.

>     * Pressing the volume (Fn+F2/F3) once changes volume twice

This is a known and reported EC bug.

febluthn, 2021-01-18:
After boot the keyboard backlight won't work at all and the tuxedo CC button does nothing (Fn + F6). If I do a suspend and wake all is working as expected again.

On the grub menu, the backlight is working.

Using Manjaro Gnome with Kernel 5.9.16-1-MANJARO.

tuxedoxt, 2021-01-18:
> Pressing the volume (Fn+F2/F3) once changes volume twice

This is now fixed in the latest EC firmware.

tuxedoxt, 2021-01-18:
> After boot the keyboard backlight won't work at all and the tuxedo CC button does nothing (Fn + F6). If I do a suspend and wake all is working as expected again.
>
> On the grub menu, the backlight is working.
>
> Using Manjaro Gnome with Kernel 5.9.16-1-MANJARO.

Regarding this issue, I believe it is not known, although the specific install has not been tested. To better track these issues please open a ticket with support.

HTOgit, 2021-01-18:
I also had the problem. Sometimes I had to re-enable the backlight in the BIOS. The problem has not occurred since the last EC update.

tuxedoxt, 2021-01-18:
> I also had the problem. Sometimes I had to re-enable the backlight in the BIOS. The problem has not occurred since the last EC update.

That's good news and good advice. Try the latest firmware. It was updated rather recently last week.

kladnik, 2021-01-18:
I updated the EC and the double volume is fixed. Side note, I downloaded the EC version 1.07.30, but in bios it is stated I have 1.14.20.

Is that correct?

tuxedoxt, 2021-01-19:
> I updated the EC and the double volume is fixed. Side note, I downloaded the EC version 1.07.30, but in bios it is stated I have > 1.14.20.
> 
> Is that correct?

Yes `1.14.20` should be the correct version. Where did it say `1.07.30`? In the downloads area?

kladnik, 2021-01-19:
> Yes `1.14.20` should be the correct version. Where did it say `1.07.30`? In the downloads area?

Yes, in there:
[image]

@tuxedoxt On other note: after upgrading EC&BIOS, my battery is not charging further than 76%, what can I do regarding this? I tried restoring BIOS settings.

tuxedoxt, 2021-01-21:
Update: The labeling should be fixed.

> @tuxedoxt On other note: after upgrading EC&BIOS, my battery is not charging further than 76%, what can I do regarding this? I tried restoring BIOS settings.

I did not manage to reproduce this. Newest firmware version, discharged and recharged fine multiple times.

kladnik, 2021-01-21:
> I did not manage to reproduce this. Newest firmware version, discharged and recharged fine multiple times.

I discharged completely (to 0%) and left it of for a bit, now it works fine again.

> When disabling the touchpad the light should be turned on (small indicator that it's off), but it never is
Once this is fixed I'll close the ticket, since it's the only thing left

kladnik, 2021-02-19:
Touchpad light works fine and also disabling/enabling is much more reliable once I installed:
https://github.com/tuxedocomputers/tuxedo-touchpad-switch

I did notice that Play/Pause with Fn key still activates twice (meaning it doesn't really pause since it pauses and instantly plays it back), but this is a minor issue that is not that important.

I guess we can close this ticket.

closed 2021-02-19 by kladnik

tuxedoxt, 2021-02-19:
> Touchpad light works fine and also disabling/enabling is much more reliable once I installed:
https://github.com/tuxedocomputers/tuxedo-touchpad-switch

Yes this was recently released. Can already be installed manually through the repos.

> Play/Pause with Fn key

Ehm I'm pretty sure this device does not have play/pause through fn-keys... at least not in our configuration.

Double volume up/down should be fixed by using the latest EC.

kladnik, 2021-02-19:
> Ehm I'm pretty sure this device does not have play/pause through fn-keys... at least not in our configuration.

It does actually, the button right bellow ESC. When I press it  it does pause, but it also instantly resumes. Not a big deal, but just wanted to let you know. 

[image]

tuxedoxt, 2021-02-19:
Ah, you are right! I had a too quick a look at a device with weird layout. Quickly tried and yes it seems to have the same affliction as the volume buttons had previously... thanks, expect another EC update, eventually.

Mi-La, 2021-03-22:
Hi. The download page now reports 1.14.20.00 for EC firmware, but when I download and unpack it, it contains 1.7.30.00.
I've faced this issue because I cannot disable my touchpad by the FN key (should be FN+F9 on Pulse 15) and also the button in left upper corner with the LED doesn't work. Do I need some special setup to make it working? I installed tuxedo-control-center together with tuxedo-keyboard, but it didn't help. Maybe I need some kernel parameter to pass it via grub?

tuxedoxt, 2021-03-23:
Hello,

I'll try to sort out some of the confusion.

- 1.7.30 is the latest EC to date (we've double checked with the ODM). I'll check the labeling of the download section though.
- For touchpad on/off software switch functionality (Fn+F9 or double tap "touchpad led location") only `tuxedo-keyboard` is necessary in most cases. However the window environment needs to support the trigger key code. With Xfce I do not know off hand.
- The extra package `tuxedo-touchpad-switch` enables hardware switch as well and the LED to show the current state. This package is a user space driver that is dependent on the window environment as well. Please see https://github.com/tuxedocomputers/tuxedo-touchpad-switch for tested systems.

Mi-La, 2021-03-23:
Thanks for the message.
Regarding the EC, I really still see:
[image]
- but it will be probably fixed.

Regarding the touchpad switch. `tuxedo-touchpad-switch` reports that my desktop environment is not supported. When I install `tuxedo-keyboard` - it seems to be working, but I found the following in dmesg:
```
$ dmesg | grep tuxedo
[    3.027069] tuxedo_keyboard: loading out-of-tree module taints kernel.
[    3.027091] tuxedo_keyboard: module verification failed: signature and/or required key missing - tainting kernel
[    3.027447] tuxedo_keyboard: Model 'TUXEDO Pulse 15 Gen1' found
[    3.027635] input: TUXEDO Keyboard as /devices/platform/tuxedo_keyboard/input/input13
```
I don't know if it's correct. However the switch still doesn't work. Do you have any hints what to check & try in XFCE? Should I inspect the source code of tuxedo-keyboard and tuxedo-touchpad-switch?
The only way I was able to disable touchpad is:
```
xinput set-prop DEVICE-ID-HERE "Device Enabled" 0
```
But not surprisingly the LED does not react to this.

tuxedoxt, 2021-03-23:
> Regarding the EC, I really still see:
> [image]
> 
>     * but it will be probably fixed.

Yeah, I do not know how this slid back to the old version but it will be fixed shortly.

If Xfce software switch (non-led) functionality does not work it probably does not respond to the same key code. Therefore if you want to find a solution for this you'd need to find out what trigger Xfce needs (if it's supported).

Others listen for the not so obvious key code `KEY_F21` which is what `tuxedo-keyboard`maps the Fn+F9 key press to. This is all it does so no need to reverse engineer this ;) If you find a solution for Xfce I'd be happy to include it.

The obvious key code to try would be `KEY_TOUCHPAD_TOGGLE` (https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h#L581) which, surprisingly is not used in general.

Mi-La, 2021-03-23:
Unfortunately I don't understand what is needed to do. When I use `xev` and press Fn+F9 or double tap to the touchpad LED button, I get the following:
```
KeyPress event, serial 39, synthetic NO, window 0xa000001,
    root 0x6b8, subw 0x0, time 51513991, (160,-9), root:(1095,450),
    state 0x10, keycode 199 (keysym 0x1008ffa9, XF86TouchpadToggle), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 39, synthetic NO, window 0xa000001,
    root 0x6b8, subw 0x0, time 51513991, (160,-9), root:(1095,450),
    state 0x10, keycode 199 (keysym 0x1008ffa9, XF86TouchpadToggle), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```
But it does nothing. I tried to modify `tuxedo-touchpad-switch` to just call `set_touchpad_state` with either 1 or 0 and  nothing more. And after I've called it with 0, I've seen the touchpad LED ON for the first time and the touchpad was disabled. So at least this works. To be honest I don't understand why there should be some integration to XFCE. It should be enough to just modify rights to allow user to call the `ioctl` (which `tuxedo-touchpad-switch` package probably does) and then bind Fn+F9 to call the toggling ioctl.

Mi-La, 2021-03-23:
Maybe this is related https://gitlab.xfce.org/xfce/xfce4-settings/-/issues/24?

Mi-La, 2021-03-24:
I just found that without `tuxed-keyboard` package I don't get the `XF86TouchpadToggle` key. So `tuxedo-keyboard` helps with getting the proper key either for Fn+F9 or double tap on touchpad LED. However it doesn't change its state. My current solution is based on `tuxedo-touchpad-switch`, I've modified touchpad-control.cpp source, changed `set_touchpad_state` to `toggle_touchpad_state` - I first read the touchpad state using ioctl, and then I swap its state using another ioctl. I made an executable from it called 'touchpad-toggle' and I was able to successfully set-up a keyboard shortcut.

But I thought that `tuxedo-keyboard` should do it without any other app and that `tuxedo-touchpad-switch` is not needed at all? So why the `tuxedo-keyboard` doesn't do its job?

tuxedoxt, 2021-03-25:
Correct, in the case of this device tuxedo-keyboard catches another sequence of key events and throws the commonly used touchpad toggle key `KEY_F21`. For most systems this is enough and leads to a software toggle + OSD from the window environment. This was found to be the most reliable way for most devices/systems and why it is not doing any "direct" toggling.

`tuxedo-touchpad-switch` is another layer that attempts to co-exist with the window manager functionality (when existent) to allow the hardware switch (incl. led) to be toggled in addition.

> Maybe this is related https://gitlab.xfce.org/xfce/xfce4-settings/-/issues/24?
> I can map the XF86TouchpadToggle to something else, e.g. using `xcape`. I understand that XF86TouchpadToggle is X key code but I don't understand to what I should map it to make `tuxedo-keyboard` working.

Seems you discovered that it's not supported by Xfce. In general touchpad toggling is not very standardised.

I'm sure there will be improvements in this area eventually and hopefully better (more standard) support for touchpad toggling for Linux in the future. In the meantime it's good that you found a workaround for your preferred system.

Mi-La, 2021-03-25:
Thanks for the confirmation. Therefore normally I would need the `tuxedo-touchpad-switch` together with `tuxed-keyboard`. Does it make sense to make a new issue in `tuxedo-touchpad-switch`? I think that I understand what the `tuxedo-touchpad-switch` does, but unfortunately it fails once it cannot detect supported desktop. I would propose to have a fallback solution, which will just start-up a daemon listening on the touchpad key and sending the appropriate toggling `ioctl`s. Or yet easier it can just set-up a keyboard shortcut in unsupported environments, or at least provide a toggling utility and let users to set-up the shortcut by themselves. I personally prefer the HW switch and since there is the indicator via LED, I don't need the software to know about switching at all.

tuxedoxt, 2021-03-25:
Well it's already known, however, an issue as feature request to track interest might be useful.

ping @Matombo

commit 89b299f6 (2022-10-10): Merge branch '60-add-lightbar-support-for-stellaris-amd-gen-4' into 'master'

Resolve "Add lightbar support for Stellaris AMD gen 4"

Closes #60

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!29