tuxedocomputers/tuxedo-keyboard #61: Licensing incompatibility between Linux kernel and these kernel modules

Conan-Kudo, 2020-12-23:
I was looking into this project's kernel modules to figure out packaging it, and I discovered that this project's license is not compatible with the Linux kernel's own license, thus the combination is not redistributable.

The Linux kernel is GPLv2 (SPDX: `GPL-2.0-only`), and the kernel modules are GPLv3+ (SPDX: `GPL-3.0-or-later`). As GPLv3 is incompatible with GPLv2 when the combination of software cannot be upgraded to GPLv3 on redistribution (as is the case with the Linux kernel), shipping this or getting this merged into the mainline kernel will not be possible without fixing the licensing of these modules first.

Please can you relicense the modules to GPLv2+ (SPDX: `GPL-2.0-or-later`) to make it compatible with the Linux kernel?

knurpht, 2020-12-23:
As a Tuxedo and long time openSUSE user I'd love to see this happen. No seperate driver downloads/installs would benefit tuxedo as well IMHO

reliant8307, 2022-06-01:
Is there any plans to merge Tuxedo modules into the mainline kernel?

jondo, 2023-01-16:
@reliant8307, yes, according to https://github.com/tuxedocomputers/tuxedo-keyboard/issues/108#issuecomment-927971620 .

closed 2023-12-11 by tuxedo-bot

salim-b, 2023-12-11:
See https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/issues/137 for a follow-up issue.