tuxedocomputers/tuxedo-keyboard #63: Conf file is not taken into account

kmscks, 2020-12-28:
Hello ! 
I own a Polaris 15 with rx 2060 and Ryzen 7 4800H. I'm running Kubuntu 20.10 with 5.8.0-33 kernel. 
But I've also tried on 20.04 resulting in the same behavior.

Editing **/etc/modprobe.d/tuxedo_keyboard.conf** doesn't seem to change anything. 
I have compiled the module and some things seem to work : for example, the default color in uniwill_keyboard.h, but not the conf file. 

**modprobe tuxedo_keyboard -c | grep options** also returns the values from the file but they don't affect the keyboard (state, mode, color etc.) 

If I can help !

tuxedoxt, 2021-01-04:
Hello,

admittedly it's slightly undocumented for now and because of further restructuring. However it should currently be possible to choose basic colors and brightness with parameters and through the correct sysfs path. For this model only the color and brightness parameters work. The other parameters are "legacy" things still present for this device although only available to other devices.

Have a look at `/sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color`. `cat color_string` should display valid values for this and should be the same as for the color parameter.

gampam2000, 2021-01-04:
ah ok, and how can I change the color? I tried in /etc/modprobe.d/tuxedo_keyboard.conf 
`options tuxedo-keyboard brightness=25 color_string=RED`
But still no effect. Or what did you mean i can choose the color with sysfs path? I also tried sudo echo 0 > /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/brightness but this does not work.
Could you give a working sample?

tuxedoxt, 2021-01-04:
Yes almost right
```
options tuxedo-keyboard brightness=25 color=RED
```
should work.

To write to sysfs you either need to do something like
```
sudo -s
echo 0 > /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/brightness
```
or
```
echo 0 | sudo tee /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/brightness
```
for the one-liner.

kmscks, 2021-01-04:
Ok Thanks for the answers ! So to sum it up, for the moment :
- brightness works as expected.
- only colors in the lookup table are validated, (BLACK RED GREEN BLUE YELLOW MAGENTA CYAN WHITE). 
- if color is set as hex value, it is not validated (fallback to default UNIWILL_COLOR_DEFAULT). 
- all other aspects are not functional for the moment (modes, custom color values).
- documentation is outdated.

Can we write colors directly to sysfs ? 
Couldn't we just test param_color against regex to validate hex color value, and at least have users being able to set their own color ? 
Thanks a lot !

gampam2000, 2021-01-04:
> Can we write colors directly to sysfs ?
Yes it works:

`sudo -s
echo "GREEN" > /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/color_string`

changed the color correctly.

tuxedoxt, 2021-04-26:
I will close this for now.

closed 2021-04-26 by tuxedoxt