tuxedocomputers/tuxedo-keyboard #64: Pulse 15 Problems

gampam2000, 2020-12-31:
I'm running the pre installed Ubuntu 20.04 and the keyboard lights are always white. I tried different configurations of the tuxedo_keyboard.conf. Also the FN+ + or FN+ - does not change the brightness as mentioned in the manual.
When the Laptop disables the screen for powersaving the lights also stay on.  
Am I doing something wrong or is this a bug?

kmscks, 2021-01-01:
Maybe edit the title of the issue if it doesn't relate to Pulse but rather Polaris. Did you compile from source ? Or was tuxedo_keyboard per-installed ?
Seems to me as the issue is similar to the one I've posted #63.

gampam2000, 2021-01-01:
It was pre-installed.  Yes it seem the same problem as #63. Kmschs do the keyboard shortcuts work with your compiles version?

kmscks, 2021-01-01:
> It was pre-installed. Yes it seem the same problem as #63. Kmschs do the keyboard shortcuts work with your compiles version?

Yes. But I'm running 20.10 and since tuxedo doesn't yet have repo for 20.10, I compiled myself. On the other hand, I don't understand why the package wouldn't work with their own pre installed distro, kind of defeats the purpose...

tuxedoxt, 2021-01-04:
Hello, yes please see the other issue for manual control and colors. I am pretty sure the key combo on this model is Fn+space are you sure that you have a Polaris?

tuxedoxt, 2021-01-04:
`Yes. But I'm running 20.10`
Also the module has been known to build and work on old to very recent kernels, this is probably not the issue. The deb package from the repo most likely works on 20.10 as well without manual build.

gampam2000, 2021-01-04:
Yes FN+Space works, thanks for the tip. But still the configuration options don't change anything. 
I have this in the /etc/modprobe.d/tuxedo_keyboard.conf
`options tuxedo-keyboard mode=0 brightness=25 color_left=0xFF0000 color_center=0x00FF00 color_right=0x0000FF`

But the the light stays all white after a reboot.

Another issue is with powersaving, how can i turn the keyboards lights off when the screen blanks? Is this somehow possible?

gampam2000, 2021-01-04:
Thanks for the info.

closed 2021-01-04 by gampam2000