tuxedocomputers/tuxedo-keyboard #66: tuxedo-keyboard not working as expected

smvdheijden, 2021-01-02:
Is my Clevo P17SM keyboard still supported? I am currently running Kubuntu 20.04 and have just installed the latest tuxedo_keyboard release, but it seems that only a few functions are working:
- When pressing any of the keys for controlling the backlight I see a message appearing in the kernel log, but nothing seems to happen. 
- I was not able to set any parameters using modprobe either.
- When changing the mode to CUSTOM, using `echo 0 >> /sys/devices/platform/tuxedo_keyboard/mode` nothing happens.
- When changing the mode to BREATHE using `echo 1 >> /sys/devices/platform/tuxedo_keyboard/mode` the backlight seems to be disabled?
- All other modes seem to be working.
- Changing brightness and/or state doesn't seem to have any effect either.

tuxedoxt, 2021-01-05:
Hello,

theoretically there should be no change in the logic to the driver for your device. Other similar keyboard backlights work. However there has been some interface changes lately. Although they should be the same it is also possible something is different in this case.

Unfortunately I do not have a test device available, would you mind sharing a dump of the acpi tables and dmesg after boot? For example in ubuntu by
```
sudo apt install acpica-tools
sudo acpidump > acpidump.out

dmesg > dmesg.txt
```

smvdheijden, 2021-01-07:
Here are the dump of the acpi tables and dmesg:
[acpidump.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/5783798/acpidump.txt)
[dmesg.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/5783791/dmesg.txt)

Additional note: When starting the laptop the keyboard backlight seems to be working, but somewhere during the boot process the backlight is switched off.

tuxedoxt, 2021-01-15:
So, this is weird. Same interface. Same logic as other devices that are known working. Driver communication works in general since some modes are working.

> Additional note: When starting the laptop the keyboard backlight seems to be working, but somewhere during the boot process the backlight is switched off.

That sounds like the point where the driver kicks in. Make sure brightness is high enough and that the color is not set to "black".

smvdheijden, 2021-01-17:
I have ended up removing tuxedo-keyboard and installing the older clevo-xsm-wmi drivers (from https://bitbucket.org/tuxedocomputers/clevo-xsm-wmi/src/master/). After this the keys for controlling the backlight are working again.

Unfortunately I've not yet been able to find the cause for the problems with the tuxedo-keyboard drivers on my laptop...

closed 2023-12-11 by tuxedo-bot