tuxedocomputers/tuxedo-keyboard #67: Missing volume key stroke events

don-philipe, 2021-01-05:
Hi,

I'm using ArchLinux on an InfinityBook Pro 14 v5. I want to customize my window manager (awesomwm) and therefore was looking for the keycodes of the mute and volume buttons. But I can't find any. I already installed the [tuxedo-keyboard](https://aur.archlinux.org/packages/tuxedo-keyboard/) package that provides the functionalities from this repo.

Recording key strokes with `xev` gives me the following output for `Fn + F3` (mute), `Fn + F6` (lower volume) and `Fn + F7` (raise volume):

```
FocusOut event, serial 36, synthetic NO, window 0x1a00001,
    mode NotifyGrab, detail NotifyAncestor

FocusIn event, serial 36, synthetic NO, window 0x1a00001,
    mode NotifyUngrab, detail NotifyAncestor

KeymapNotify event, serial 36, synthetic NO, window 0x0,
    keys:  0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   
           0   0   0   0   0   0   0   0   0   0   0   0   0   0   0   0
```

I was expecting some keycode etc. like `XF86AudioMute` and so on.

Is there any other way to get the key stroke events or could this be an issue of the keyboard driver?

tuxedoxt, 2021-02-04:
Hello, `tuxedo-keyboard` currently does not provide any functionality to the volume keys. Off-hand I do not know how it is handled on your device. If you are just looking for keycodes you can try `showkey`.

wolass, 2021-03-29:
Dear Tuxedo Team, 
I am having the same issue on my Aura 15.
OS: Arch,, 
WM: AWESOME,  

The xev codes are missing for XF86AudioVolumeRise, and Lower, as well as screen backlight +/-, among others. But these are the most crucial.

My Showkey shows the  code for the key that is also the same one when I run `xmodmap -pke` . Unfortunately binding 
```
xmodmap -pke >> .Xmodmap
xmodmap .Xmodmap
```
does not produce the expected effect - xev does not recognise these keys... 

The modprobe says that tuxedo-keyboard is loaded and is version 3.0.5 
Also running various functions on xmodmap (like `mode=1` ) does not produce the expected effect i.e. changing the keyboard backlight effect to "breathe"... I also tried changing backlight colors but without any luck. 

Is there anything that could be grabbing my keys in between kernel and X session? 

What do you suggest to try? 

I would really like to have the functionality of these keys as it makes it easier to have the same media keybindings across different machines"

tuxedoxt, 2021-03-29:
Alright, as far as I recall there are no known issues for these devices related to the volume keys. Let's check the kernel command lines.

`cat /proc/cmdline`

Try removing any `osi` related parameters and any acpi backlight related ones for backlight issues.

> The modprobe says that tuxedo-keyboard is loaded and is version 3.0.5

That's weird, latest release to date is `3.0.4`.

don-philipe, 2021-03-29:
Just recently I found out that when I addionally press `Ctrl` ( so complete key combinations are `Ctrl + Fn + F3/F5/F6`) AwesomeWM recognizes these keys correctly as `XF86Audio...` keys, but `xev` still doesn't. `tuxedo-keyboard` is version 3.0.5  for me as well.

don-philipe, 2021-04-18:
I really have to apologize that I opened this issue. When I got this Tuxedo laptop, I copied an existing system to it. On this system I had the `xbindkeys` command in my `.xinitrc` that I needed on my previous laptop. It was years ago when I configured it and I totally forgot about that part.

Long story short: removing `xbindkeys` from `.xinitrc` brought back the `XF86Audio...` key events.

Sorry again for this useless issue.

closed 2021-04-18 by don-philipe

tuxedoxt, 2021-04-19:
Great, mystery solved!