tuxedocomputers/tuxedo-keyboard #68: Pulse 15 keyboard backlight does not work until until 1st wake up from suspend

febluthn, 2021-01-18:
After boot the keyboard backlight won't work at all. If I do a suspend and wake all is working as expected again.

On the grub menu, the backlight is working.
Furthermore the Tuxedo-CC button (Fn + F6) does also nothing until wake up from suspend.

Using Manjaro Gnome with Kernel 5.9.16-1-MANJARO.

what I can see on first look is:
```
$ journalctl -p 3 -xb

Jan 18 10:22:08 <hostname> kernel: ACPI BIOS Error (bug): Failure creating named object [\SMIB], AE_ALREADY_EXISTS (20200717/dsfield-637)
Jan 18 10:22:08 <hostname> kernel: ACPI BIOS Error (bug): Could not resolve symbol [\_SB.PCI0.GPP4.WLAN], AE_NOT_FOUND (20200717/dswload2-162)
Jan 18 10:22:08 <hostname> kernel: ACPI Error: AE_NOT_FOUND, During name lookup/catalog (20200717/psobject-220)
Jan 18 10:22:08 <hostname> kernel: sp5100-tco sp5100-tco: Watchdog hardware is disabled
Jan 18 10:22:08 <hostname> kernel: snd_pci_acp3x 0000:04:00.5: Invalid ACP audio mode : 0
Jan 18 10:22:13 <hostname> systemd-backlight[1124]: Failed to get backlight or LED device 'backlight:acpi_video0': No such device
Jan 18 10:22:13 <hostname> systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:acpi_video0.
	at Subject: A start job for unit systemd-backlight@backlight:acpi_video0.service has failed
	at Defined-By: systemd
	at Support: https://forum.manjaro.org/c/support
	at 
	at A start job for unit systemd-backlight@backlight:acpi_video0.service has finished with a failure.
	at 
	at The job identifier is 922 and the job result is failed.
Jan 18 10:22:13 <hostname> systemd-backlight[1125]: amdgpu_bl0: Failed to write system 'brightness' attribute: No such device or address
Jan 18 10:22:13 <hostname> systemd[1]: Failed to start Load/Save Screen Backlight Brightness of backlight:amdgpu_bl0.
	at Subject: A start job for unit systemd-backlight@backlight:amdgpu_bl0.service has failed
	at Defined-By: systemd
	at Support: https://forum.manjaro.org/c/support
	at 
	at A start job for unit systemd-backlight@backlight:amdgpu_bl0.service has finished with a failure.
	at 
	at The job identifier is 928 and the job result is failed.
```

tuxedoxt, 2021-01-18:
> After boot the keyboard backlight won't work at all. If I do a suspend and wake all is working as expected again.
> 
> On the grub menu, the backlight is working.
> Furthermore the Tuxedo-CC button (Fn + F6) does also nothing until wake up from suspend.

Hello,

as stated in the other issue, please open a ticket with support to better track this issue.

As for the messages I do not see anything relevant to keyboard backlight.

tlvlp, 2021-01-27:
I have the same issues (and a few more regarding FN key bindings) with the Pulse 14 under ubuntu gnome.

tuxedoxt, 2021-01-27:
> I have the same issues (and a few more regarding FN key bindings) with the Pulse 14 under ubuntu gnome.

What kind of Fn key bindings issues? Please see #60, you might need the latest firmware. Most issues such as those stated in that issue and eg. wrongly mapped Fn-keys will be fixed by firmware update (bios&ec).

tlvlp, 2021-01-27:
> What kind of Fn key bindings issues? Please see #60, you might need the latest firmware. Most issues such as those stated in that issue and eg. wrongly mapped Fn-keys will be fixed by firmware [...]

The latest update seems to have solved this issue (I have assumed that a laptop delivered yesterday would already come with the latest firmware).

But I still have anomalies with the upper-right keys after the F-keys:
- the 'pause' key fires the 'Print'(screen) action
- the 'prtsc' key fires the 'Insert' action
- The upper-right 'insert' key fires the 'Delete' action, which is somewhat fortunate as there is no delete key on this eng-us keyboard for some reason and I need that more than the insert action which I never use.

This is btw a desirable configuration for the above mentioned reason so is it possible that the buttons have been mislabeled during manufacturing?

tuxedoxt, 2021-01-27:
Hmm yeah if you can take a picture of the keyboard I will verify. This would be an official support matter then in this case.

tlvlp, 2021-01-27:
> Hmm yeah if you can take a picture of the keyboard I will verify. 
Sure, I have [uploaded it to my drive for now](https://drive.google.com/file/d/1rbTmQTzYiDRQnJ3ndJZ5bPBOKmKwo6mj/view?usp=sharing).

> This would be an official support matter then in this case though.
Should I send a separate email to the support address?

- In the meantime I have also discovered that the problem came back, where the backlight illumination disappears no matter what I do, until I log out and log back again. At this point I cannot yet link it to any particular action, but I assume the idle screen-lock could be the culprit.

- The key next to the left-shift also seems to be weird as it displays the key for ' \ ' and ' | ' the same as the one that is left to the 'enter' key but instead fires '<' and '>' (with alt)

tuxedoxt, 2021-01-27:
Right, yeah sorry, this is clearly not correct.

Yes, please open a ticket with support "the normal way" to get this fixed, they'll know the correct procedure.

Simply fill out the form to get it started, a brief description should suffice: https://www.tuxedocomputers.com/en/Contact.tuxedo
You should get a mail which you can answer and attach the photo, should be clear enough.

As for the irregular backlight I have not been able to reproduce it. You can also mention the keyboard backlight issue to support to get it tracked properly.

tlvlp, 2021-01-27:
> Simply fill out the form to get it started, a brief description should suffice: https://www.tuxedocomputers.com/en/Contact.tuxedo

Thank you, just did.

tuxedoxt, 2021-02-04:
I will close this for now. Feel free to open if problems issue title problem persists after bios & ec update.

closed 2021-02-04 by tuxedoxt

thepasto, 2021-03-12:
I'm having the same problem, keyboard backlight is not working... it will start to work only after a suspend. i'm running arch linux kernel v. 5.11.5-arch1-1, bios and EC are updated. Suggestions? Thanks

febluthn, 2021-03-12:
Since ec update I can just toggle backlight with fn-space. There is no automation like dim down after couple of seconds non-typing anymore. I took it as intended and do not reopen the ticket.