tuxedocomputers/tuxedo-keyboard #69: Keyboard brightness stopped working on Polaris with 5.10

ruffson, 2021-02-01:
Hey there,

I don't know if it was the new kernel version or something else but the keyboard backlighting stopped working on my Polaris 15. I updated the `tuxedo-keyboard` to version 3.0.2 but still nothing. Everything else about the keyboard seems to work except presing `FN + SPACE` to turn it on. I tried with a config file as well and set it to `255` but that did not do anything else.

My system: 
Fedora 33, Gnome `3.38.3`, X11, kernel `5.10.10-200`

I also tried uninstalling the dkms module with `dkmsremove` but even then no luck (when I initially installed Fedora, the keyboard worked fine including backlight and backlight control, just the color was random and disabling the trackpad did not work, so I went ahead and installed it)

Also when I reboot the computer, the trackpad does not always work and requires a second reboot after which it does. That behaviour is also new. The backlight still works in general, because when booting up it lights up and goes through a color cycle.

Do you guys have any idea what the problem could be here?

lldb3, 2021-02-04:
Hi, I have the same system and similar specs so maybe we can debug sth out, only difference is I am using Wayland, but everything I am mentioning also applies to X11. 
* Fedora 33
* GNOME 3.38.3
* **Kernel 5.10.11-200.fc33.x86_64**  This one is different, maybe try it. I have other problems but can at least cycle brightness..

For me the backlight works with `Fn+Space` and cycles through levels of  brightness. But any other feature is not present (parameters inside sublevels of `/sys/devices/platform/tuxedo_keyboard' are limited to brightness and color_string). 

Modprobe params are: `options tuxedo-keyboard mode=0 brightness=25 color_left=0x0000FF color_center=0x00FF00 color_right=0xFF0000`, yet this has no effect, I only get white color with no brightness (works in bootloader though). Will try again with different params tonight.

> Also when I reboot the computer, the trackpad does not always work and requires a second reboot after which it does. That behaviour is also new. The backlight still works in general, because when booting up it lights up and goes through a color cycle.

This is one I experience as well. The machine is new so I'm glad to see this is not a manufacturing defect haha. When I have time I will investigate a bit further..

ruffson, 2021-02-05:
Gladly! I have to add that originally the Polaris I received worked fine out of the box with Fedora and `5.9`. I received a replacement model and just used my SSDs I previously had in the old model. I think initially the lighting worked but eventually just stopped. And I think that correlated with `5.10`. I don't know any other major system change. 

In any case it would be good if we had a way of diagnosing the problem.

tuxedoxt, 2021-02-05:
Hello, which Polaris models do you have here?
```
cat /sys/class/dmi/id/board_name
```
should provide our model ID.

At least one Polaris model has been known to work with the ubuntu oem-20.04-edge (5.10) kernel.

Here is another issue (https://github.com/tuxedocomputers/tuxedo-keyboard/issues/63) related to backlight since I believe there is still no explicit documentation on writing the parameters yourself. (They are slightly different than classical tuxedo-keyboard.)

ruffson, 2021-02-05:
> Hello, which Polaris models do you have here?
> 
> ```
> cat /sys/class/dmi/id/board_name
> ```
> 
> should provide our model ID.

For me it just says:
```
> cat /sys/class/dmi/id/board_name
Standard
```
I have the Polaris Gen1 with AMD Ryzen 4800H, RTX 2060.

lldb3, 2021-02-05:
> Hello, which Polaris models do you have here?
> 
> ```
> cat /sys/class/dmi/id/board_name
> ```

```shell
$ cat /sys/class/dmi/id/board_name  
POLARIS1701A2060
```
I have a Polaris 17 inch Gen1 with AMD Ryzen 4800H, RTX 2060.

tuxedoxt, 2021-02-05:
> For me it just says:
> 
> ```
> > cat /sys/class/dmi/id/board_name
> Standard
> ```
> 
> I have the Polaris Gen1 with AMD Ryzen 4800H, RTX 2060.

Ok this is not correct. It will need a firmware update for correct identification. Please download the firmware for your device from the download area in your account and update according to the instructions on
 https://www.tuxedocomputers.com/en/Infos/Help-Support/Instructions/Updating-EFI-BIOS-EC.tuxedo.

lldb3, 2021-02-05:
> Here is another issue (#63) related to backlight since I believe there is still no explicit documentation on writing the parameters yourself. (They are slightly different than classical tuxedo-keyboard.)

This worked indeed, thanks !

ruffson, 2021-02-05:
> Ok this is not correct. It will need a firmware update for correct identification. Please download the firmware for your device from the download area in your account and update according to the [...]

Ah that makes sense. It came to me with the default AMI BIOS and I only updated that and not the EC because the version seemed up-to-date. Will then try to update the EC as well later. Thank you!

tuxedoxt, 2021-02-05:
> Ah that makes sense. It came to me with the default AMI BIOS and I only updated that and not the EC because the version seemed up-to-date. Will then try to update the EC as well later. Thank you!

That should do it, let me know how it works.

ruffson, 2021-02-05:
That fixed it, thank you!

commit 7c8098b8 (2023-03-28): Merge branch 'fan-ctl-id-tweak' into 'master'

Fan ctl id tweak

Closes #69

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!41
closed 2021-02-05 by ruffson