tuxedocomputers/tuxedo-keyboard #71: Hardware not updating PowerSpec 1530 (Clevo PB50DF2)

m-wells, 2021-02-04:
My keyboard backlight works in Windows 10 and remains blue in Arch Linux (kernel 5.10.12-arch1-1). The hardware keys do nothing to change the lighting of my keyboard although the system configuration seems to be updating. Here is the result of `journalctl` after hitting `Fn` and the numpad `-`, `+`, `/`, and `*` multiple times in no particular order. Again there is no change in brightness or color of the keyboard during any of this. The `atkbd serio0` messages occur when I press and then again when I release the `Fn` key only.
```
Feb 04 12:25:50 beryllium kernel: tuxedo_keyboard: Set brightness on 75
Feb 04 12:25:50 beryllium kernel: tuxedo_keyboard: Set brightness on 50
Feb 04 12:25:51 beryllium kernel: tuxedo_keyboard: Set brightness on 25
Feb 04 12:25:51 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:25:51 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:25:51 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:25:51 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:25:52 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:25:52 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:25:52 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:04 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:04 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:06 beryllium kernel: tuxedo_keyboard: Set keyboard enabled to: 0
Feb 04 12:29:07 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:07 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:07 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:07 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:08 beryllium kernel: tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 2, new_color_code FF00
Feb 04 12:29:08 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:08 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:09 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:09 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:11 beryllium kernel: tuxedo_keyboard: Set brightness on 0
Feb 04 12:29:11 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:11 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:12 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:12 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:12 beryllium kernel: tuxedo_keyboard: Set brightness on 25
Feb 04 12:29:12 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:12 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:42 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:42 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:44 beryllium kernel: tuxedo_keyboard: Set brightness on 50
Feb 04 12:29:44 beryllium kernel: tuxedo_keyboard: Set brightness on 75
Feb 04 12:29:44 beryllium kernel: tuxedo_keyboard: Set brightness on 100
Feb 04 12:29:44 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:44 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:45 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:45 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:45 beryllium kernel: tuxedo_keyboard: Set brightness on 125
Feb 04 12:29:46 beryllium kernel: tuxedo_keyboard: Set brightness on 150
Feb 04 12:29:46 beryllium kernel: tuxedo_keyboard: Set brightness on 175
Feb 04 12:29:46 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:46 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:46 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:46 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:47 beryllium kernel: tuxedo_keyboard: Set keyboard enabled to: 1
Feb 04 12:29:47 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:47 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:47 beryllium kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:47 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Feb 04 12:29:48 beryllium kernel: tuxedo_keyboard: set_next_color_whole_kb(): new_color_id: 3, new_color_code FF
Feb 04 12:29:48 beryllium kernel: atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Feb 04 12:29:48 beryllium kernel: atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
```
[acpidump.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/5927470/acpidump.txt)
[dmesg.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/5927474/dmesg.txt)

Some more detailed system info (with serial numbers and uuids removed).
```
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Micro Electronics Inc
	Product Name: 1530
	Version: Not Applicable
	Wake-up Type: Power Switch
	Family: PowerSpec

Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
	Manufacturer: Clevo
	Product Name: PB50DF2
	Version: Not Applicable
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: To Be Filled By O.E.M.
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
```

tuxedoxt, 2021-02-04:
Hello,

looks like you have USB keyboard backlight (`ITE Tech. Inc. ITE Device(829x)`).

This is not yet included in the `tuxedo-keyboard` package. Try `tuxedo-keyboard-ite` in combination, it might have a basic driver for your device.

m-wells, 2021-02-04:
Wow. Like a charm. This has been driving me crazy.

closed 2021-02-05 by tuxedoxt