tuxedocomputers/tuxedo-keyboard #72: tuxedo-keyboard : Conflicts: tuxedo-cc-wmi but 0.1.7 is to be installed

JanekNaymspace, 2021-02-04:
When trying to upgrade from `tuxedo-keyboard 2.1.0` to >= 3.0.0, the title message appears.

```
$ dpkg --list | grep tuxedo
ii  tuxedo-cc-wmi                                 0.1.7                                       all          Interface to WMI methods/control on TUXEDO Laptops
ii  tuxedo-control-center                         1.0.7                                       amd64        
ii  tuxedo-keyboard                               2.1.0                                       all          Keyboard driver for TUXEDO notebooks.
```

```
$ sudo apt upgrade tuxedo-keyboard
The following packages have unmet dependencies:
 tuxedo-keyboard : Conflicts: tuxedo-cc-wmi but 0.1.7 is to be installed
E: Broken packages
```

Do I have to provide further information?

tuxedoxt, 2021-02-05:
Hello, to update only a single package
```
sudo apt install tuxedo-keyboard
```
should work fine and tell you that `tuxedo-cc-wmi` will be removed since it is now part of `tuxedo-keyboard`.

JanekNaymspace, 2021-02-05:
Thanks for this hint – works great :+1:

closed 2021-02-05 by JanekNaymspace