tuxedocomputers/tuxedo-keyboard #73: No rule to make target 'clean'

Gene-W, 2021-02-05:
Hello,
I followed the instructions for git-clone and then jumped to DKMS instructions. As soon as I enter make clean i receive the following:

```
make -C /lib/modules/5.9.16-1-MANJARO/build M=/home/irish/tuxedo-keyboard clean
make[1]: Entering directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make[1]: *** No rule to make target 'clean'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/5.9.16-1-MANJARO/build'
make: *** [Makefile:31: clean] Error 2
```

I  did some searching and there seem to be quite a lot of suggestions for similar yet different situations. I did try some of the suggestions (like 'make -f Makefile clean) but nothing changed the error output, so I thought I would ask for a little help. Thank you!

Manjaro 20.2.1
Gnome 3.38.3
Kernel 5.9.16-1
Clevo N85_N87,HJ,HJ1,HK1 (dmidecode)

tuxedoxt, 2021-02-05:
Hello,

it depends a little on what you are intending to do. If you have not built it yet you do not need to clean it. For each kernel you intend to build for (with dkms) you do need to install the kernel headers though.

Gene-W, 2021-02-05:
Thank you for your reply. My attempt is to get my keyboard backlights working and I was following the DKMS route instructions. I checked and see that I have this after upgrading headers: 
```
pacman -Q | grep headers
linux-api-headers 5.8-1
linux59-headers 5.9.16-1
```

Does that mean I jump straight to `sudo make dkmsinstall` and follow from there? My apologies for what are probably quite basic questions. I am comfortable in the environment but don't always know the way forward. :)

tuxedoxt, 2021-02-05:
If you are just looking to get it working I would recommend searching an already packaged option for manjaro.

Unfortunately I am not familiar enough with how the kernels in manjaro are packaged t ogive a definite answer.

Somehow it sounds like it might be installed. Yes you could try `sudo make dkmsinstall` it will tell you rather clearly if it managed to find headers and build it or not. Still I recommend the prepackaged options.

Gene-W, 2021-02-05:
> If you are just looking to get it working I would recommend searching an already packaged option for manjaro.

Sure enough, in the repository is tuxedo-keyboard 3.0.1-2 and it all worked out fast and easy. I just didn't check there first! Your help is very much appreciated, thank you!

closed 2021-02-06 by Gene-W