tuxedocomputers/tuxedo-keyboard #74: Full system freeze

m-wells, 2021-02-08:
When I modify the backlight using the hardware keys I get the following output from `dmesg`
```
[  457.803575] BUG: workqueue leaked lock or atomic: kworker/0:0/0x7ffffffe/3970
                    last function: console_callback
[  457.803578] CPU: 0 PID: 3970 Comm: kworker/0:0 Tainted: P        W  OE     5.10.14-arch1-1 #1
[  457.803580] Hardware name: Micro Electronics Inc 1530/PB50DF2, BIOS 1.07.06TIP 09/25/2019
[  457.803583] Workqueue: events console_callback
[  457.803586] Call Trace:
[  457.803592]  dump_stack+0x6b/0x83
[  457.803597]  process_one_work.cold+0x2b/0x30
[  457.803604]  worker_thread+0x4d/0x3d0
[  457.803609]  ? rescuer_thread+0x3c0/0x3c0
[  457.803614]  kthread+0x133/0x150
[  457.803618]  ? __kthread_bind_mask+0x60/0x60
[  457.803624]  ret_from_fork+0x1f/0x30
[  457.803630] BUG: scheduling while atomic: kworker/0:0/3970/0x7fffffff
[  457.803632] Modules linked in: rfcomm cmac algif_hash algif_skcipher af_alg snd_usb_audio razermouse(OE) razeraccessory(OE) snd_usbmidi_lib snd_rawmidi snd_seq_device tuxedo_keyboard_ite(OE) bnep btusb btrtl btbcm btintel bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev ecdh_generic usbhid mc ecc ccm nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) tpm_crb snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic joydev mousedev hid_multitouch intel_rapl_msr intel_rapl_common iTCO_wdt mei_hdcp ee1004 intel_pmc_bxt iTCO_vendor_support wmi_bmof intel_wmi_thunderbolt clevo_wmi(OE) snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda x86_pkg_temp_thermal snd_sof intel_powerclamp ledtrig_audio iwlmvm coretemp kvm_intel snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core mac80211 snd_soc_acpi_intel_match snd_soc_acpi snd_hda_intel snd_intel_dspcfg soundwire_intel
[  457.803727]  soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_hda_core kvm libarc4 i915 snd_hwdep irqbypass soundwire_bus crct10dif_pclmul iwlwifi crc32_pclmul ghash_clmulni_intel snd_soc_core aesni_intel i2c_algo_bit crypto_simd cryptd drm_kms_helper glue_helper snd_compress rapl ac97_bus nls_iso8859_1 intel_cstate snd_pcm_dmaengine r8169 vfat intel_uncore cfg80211 pcspkr snd_pcm cec fat psmouse realtek mdio_devres snd_timer i2c_i801 intel_gtt libphy i2c_smbus syscopyarea snd mei_me sysfillrect intel_lpss_pci sysimgblt intel_lpss thunderbolt soundcore mei i2c_nvidia_gpu rfkill fb_sys_fops idma64 intel_pch_thermal i2c_hid tpm_tis clevo_acpi(OE) video tpm_tis_core tuxedo_io(OE) tuxedo_keyboard(OE) tpm wmi intel_hid mac_hid rng_core sparse_keymap acpi_pad drm i2c_dev fuse crypto_user agpgart bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 rtsx_pci_sdmmc mmc_core crc32c_intel serio_raw rtsx_pci xhci_pci xhci_pci_renesas
[  457.803846] CPU: 0 PID: 3970 Comm: kworker/0:0 Tainted: P        W  OE     5.10.14-arch1-1 #1
[  457.803848] Hardware name: Micro Electronics Inc 1530/PB50DF2, BIOS 1.07.06TIP 09/25/2019
[  457.803852] Workqueue:  0x0 (events)
[  457.803857] Call Trace:
[  457.803864]  dump_stack+0x6b/0x83
[  457.803870]  __schedule_bug.cold+0x4c/0x58
[  457.803875]  __schedule+0x6d7/0x810
[  457.803881]  schedule+0x5b/0xc0
[  457.803887]  worker_thread+0xbf/0x3d0
[  457.803893]  ? rescuer_thread+0x3c0/0x3c0
[  457.803898]  kthread+0x133/0x150
[  457.803902]  ? __kthread_bind_mask+0x60/0x60
[  457.803908]  ret_from_fork+0x1f/0x30
[  457.803922] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #40!!!
[  458.060553] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
[  458.060559] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[  458.936410] atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
[  458.936417] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[  459.347404] tuxedo_keyboard: Set brightness on 0
[  459.356194] BUG: scheduling while atomic: kworker/0:0/3970/0x00000003
[  459.356199] Modules linked in: rfcomm cmac algif_hash algif_skcipher af_alg snd_usb_audio razermouse(OE) razeraccessory(OE) snd_usbmidi_lib snd_rawmidi snd_seq_device tuxedo_keyboard_ite(OE) bnep btusb btrtl btbcm btintel bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev ecdh_generic usbhid mc ecc ccm nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) tpm_crb snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic joydev mousedev hid_multitouch intel_rapl_msr intel_rapl_common iTCO_wdt mei_hdcp ee1004 intel_pmc_bxt iTCO_vendor_support wmi_bmof intel_wmi_thunderbolt clevo_wmi(OE) snd_sof_pci snd_sof_intel_byt snd_sof_intel_ipc snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_xtensa_dsp snd_sof_intel_hda x86_pkg_temp_thermal snd_sof intel_powerclamp ledtrig_audio iwlmvm coretemp kvm_intel snd_soc_skl snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core mac80211 snd_soc_acpi_intel_match snd_soc_acpi snd_hda_intel snd_intel_dspcfg soundwire_intel
[  459.356310]  soundwire_generic_allocation soundwire_cadence snd_hda_codec snd_hda_core kvm libarc4 i915 snd_hwdep irqbypass soundwire_bus crct10dif_pclmul iwlwifi crc32_pclmul ghash_clmulni_intel snd_soc_core aesni_intel i2c_algo_bit crypto_simd cryptd drm_kms_helper glue_helper snd_compress rapl ac97_bus nls_iso8859_1 intel_cstate snd_pcm_dmaengine r8169 vfat intel_uncore cfg80211 pcspkr snd_pcm cec fat psmouse realtek mdio_devres snd_timer i2c_i801 intel_gtt libphy i2c_smbus syscopyarea snd mei_me sysfillrect intel_lpss_pci sysimgblt intel_lpss thunderbolt soundcore mei i2c_nvidia_gpu rfkill fb_sys_fops idma64 intel_pch_thermal i2c_hid tpm_tis clevo_acpi(OE) video tpm_tis_core tuxedo_io(OE) tuxedo_keyboard(OE) tpm wmi intel_hid mac_hid rng_core sparse_keymap acpi_pad drm i2c_dev fuse crypto_user agpgart bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 rtsx_pci_sdmmc mmc_core crc32c_intel serio_raw rtsx_pci xhci_pci xhci_pci_renesas
[  459.356446] CPU: 0 PID: 3970 Comm: kworker/0:0 Tainted: P        W  OE     5.10.14-arch1-1 #1
[  459.356448] Hardware name: Micro Electronics Inc 1530/PB50DF2, BIOS 1.07.06TIP 09/25/2019
[  459.356458] Workqueue: kacpi_notify acpi_os_execute_deferred
[  459.356461] Call Trace:
[  459.356476]  dump_stack+0x6b/0x83
[  459.356484]  __schedule_bug.cold+0x4c/0x58
[  459.356490]  __schedule+0x6d7/0x810
[  459.356496]  schedule+0x5b/0xc0
[  459.356502]  schedule_timeout+0x98/0x160
[  459.356510]  ? __next_timer_interrupt+0x100/0x100
[  459.356515]  wait_for_completion_timeout+0x99/0x110
[  459.356522]  usb_start_wait_urb+0xa8/0x190
[  459.356528]  usb_control_msg+0xe7/0x150
[  459.356538]  usbhid_raw_request+0x99/0x160 [usbhid]
[  459.356545]  keyb_send_data.isra.0+0xa2/0xe0 [tuxedo_keyboard_ite]
[  459.356551]  keyboard_notifier_callb+0xac/0x190 [tuxedo_keyboard_ite]
[  459.356559]  atomic_notifier_call_chain+0x51/0x80
[  459.356566]  kbd_event+0x263/0x780
[  459.356572]  ? __queue_work+0x1dd/0x420
[  459.356578]  input_to_handler+0xd4/0xf0
[  459.356585]  input_pass_values.part.0+0x126/0x140
[  459.356591]  input_handle_event+0x101/0x5f0
[  459.356596]  input_event+0x53/0x80
[  459.356602]  sparse_keymap_report_entry+0x9b/0xc0 [sparse_keymap]
[  459.356608]  clevo_acpi_notify+0x55/0x90 [clevo_acpi]
[  459.356614]  acpi_ev_notify_dispatch+0x4a/0x5f
[  459.356620]  acpi_os_execute_deferred+0x16/0x20
[  459.356626]  process_one_work+0x1d6/0x3a0
[  459.356632]  worker_thread+0x4d/0x3d0
[  459.356638]  ? rescuer_thread+0x3c0/0x3c0
[... 37 lines cut ...]
[  459.356870]  worker_thread+0x4d/0x3d0
[  459.356876]  ? rescuer_thread+0x3c0/0x3c0
[  459.356880]  kthread+0x133/0x150
[  459.356885]  ? __kthread_bind_mask+0x60/0x60
[  459.356890]  ret_from_fork+0x1f/0x30
[  459.356896] BUG: scheduling while atomic: kworker/0:0/3970/0x7fffffff
[  459.356898] Modules linked in: rfcomm cmac algif_hash algif_skcipher af_alg snd_usb_audio razermouse(OE) razeraccessory(OE) snd_usbmidi_lib snd_rawmidi snd_seq_device tuxedo_keyboard_ite(OE) bnep btusb btrtl btbcm btintel bluetooth uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev ecdh_generic usbhid mc ecc ccm nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) tpm_crb snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic joydev mousedev hid_multitouch intel_rapl_msr intel_rapl_common iTCO_wdt mei_hdcp ee1004 intel_pmc_bxt iTCO_vendor_support wmi_bmof intel_wmi_thunderbolt clevo_w [... cut ...]
[  459.357153]  worker_thread+0xbf/0x3d0
[  459.357159]  ? rescuer_thread+0x3c0/0x3c0
[  459.357164]  kthread+0x133/0x150
[  459.357168]  ? __kthread_bind_mask+0x60/0x60
[  459.357174]  ret_from_fork+0x1f/0x30
[  459.669691] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
[  459.669699] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
```
While my backlight works, if I change the settings too many times I get a full system freeze (can't even change to a different `tty`).

tuxedoxt, 2021-02-09:
Hello,

how many times are too many? Could you provide more specific instruction for reproducing? Special combination, timing, anything?

m-wells, 2021-02-09:
Well uninstalling `tuxedo-keyboard-ite` fixes the system freeze issue. This also means that I can no longer control the backlight. This issue is incredibly frustrating to debug as it locks up my system each time. I can say that pressing any combination of keyboard backlight keys in rapid succession causes this issue. Such as going from minimum brightness to max or cycling display modes.

tuxedoxt, 2021-02-10:
> Well uninstalling `tuxedo-keyboard-ite` (@StevenSeifried) fixes the system freeze issue. This also means that I can no longer control the backlight. This issue is incredibly frustrating to debug as it locks up my system each time. I can say that pressing any combination of keyboard backlight keys in rapid succession causes this issue. Such as going from minimum brightness to max or cycling display modes.

Thanks, I'll look into it.

tuxedoxt, 2021-02-18:
Hello,

a new version of `tuxedo-keyboard-ite` should be available (v0.2.2) with a possible fix.

Appreciate a short feedback whether this fixes your issue or not.

m-wells, 2021-02-19:
Awesome! No crash and I can control the backlight again. One thing is the brightness on the keyboard itself is not changing. I can change the mode and the state (so toggle it on or off). But the actual brightness remains constant. Even going down to brightness 0 the keyboard remains illuminated.

The output of `dmesg -w`
```
[ 4786.780721] tuxedo_keyboard: Set brightness on 0
[ 4787.050532] tuxedo_keyboard: Set brightness on 0
[ 4787.277382] tuxedo_keyboard: Set brightness on 0
[ 4787.483618] tuxedo_keyboard: Set brightness on 0
[ 4787.665472] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
[ 4787.665484] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[ 4787.860535] atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
[ 4787.860547] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[ 4788.123856] tuxedo_keyboard: Set brightness on 25
[ 4788.377168] tuxedo_keyboard: Set brightness on 50
[ 4788.603676] tuxedo_keyboard: Set brightness on 75
[ 4788.796749] tuxedo_keyboard: Set brightness on 100
[ 4789.128584] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
[ 4789.128599] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[ 4789.331532] atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
[ 4789.331541] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
[ 4789.550270] tuxedo_keyboard: Set brightness on 125
[ 4789.776614] tuxedo_keyboard: Set brightness on 150
[ 4790.006869] tuxedo_keyboard: Set brightness on 125
[ 4790.213702] tuxedo_keyboard: Set brightness on 100
[ 4790.460264] tuxedo_keyboard: Set brightness on 125
[ 4790.656942] tuxedo_keyboard: Set brightness on 150
[ 4790.836652] tuxedo_keyboard: Set brightness on 125
[ 4791.051062] tuxedo_keyboard: Set brightness on 100
[ 4791.273820] tuxedo_keyboard: Set brightness on 125
[ 4791.446533] tuxedo_keyboard: Set brightness on 150
```
The Unknown key released is when I press and release `Fn`.

Being able to actually toggle the state of the keyboard is a great improvement and very much appreciated.

tuxedoxt, 2021-02-22:
Glad the crashes are gone!

Hmm, you are right, the brightness somehow broke in the process. New version is on the way, probably available sometime tomorrow.

tuxedoxt, 2021-02-24:
New version of `tuxedo-keyboard-ite` (v0.2.3) should now be in our repos.

m-wells, 2021-02-24:
This seems to have done the trick! Thanks for addressing this issue so quickly.

closed 2021-02-25 by vinzv