tuxedocomputers/tuxedo-keyboard #75: sysfs: cannot create duplicate filename '/devices/platform/tuxedo_keyboard'

ftufo, 2021-02-10:
After updating to the latest Xubuntu 20.04 kernel (5.10.0-1013-oem), I get the following error on boot. It was present also on previous 5.10.* kernels, but now it is deterministic (at every boot). Never saw it on 5.8.* kernels.
By the way, everything seems to work OK.

```
[    4.229219] sysfs: cannot create duplicate filename '/devices/platform/tuxedo_keyboard'
[    4.229222] CPU: 3 PID: 364 Comm: systemd-udevd Tainted: G           OE     5.10.0-1013-oem #14-Ubuntu
[    4.229223] Hardware name: TUXEDO TUXEDO Aura 15 Gen1/AURA1501, BIOS 1.07.03RTR2 11/13/2020
[    4.229224] Call Trace:
[    4.229230]  dump_stack+0x74/0x92
[    4.229232]  sysfs_warn_dup.cold+0x17/0x27
[    4.229234]  sysfs_create_dir_ns+0xb8/0xd0
[    4.229237]  kobject_add_internal+0xbd/0x2b0
[    4.229238]  kobject_add+0x7e/0xb0
[    4.229241]  ? kmem_cache_alloc_trace+0x37a/0x430
[    4.229243]  device_add+0x132/0x870
[    4.229245]  platform_device_add+0xf9/0x240
[    4.229247]  __platform_create_bundle+0x9f/0xe0
[    4.229249]  ? set_color+0x80/0x80 [tuxedo_keyboard]
[    4.229251]  ? clevo_wmi_notify+0x80/0x80 [clevo_wmi]
[    4.229252]  tuxedo_keyboard_init_driver+0x51/0x1c0 [tuxedo_keyboard]
[    4.229254]  clevo_keyboard_init+0x15/0x50 [tuxedo_keyboard]
[    4.229255]  clevo_wmi_probe+0x103/0x120 [clevo_wmi]
[    4.229258]  wmi_dev_probe+0x187/0x1d0 [wmi]
[    4.229259]  really_probe+0x1db/0x440
[    4.229260]  driver_probe_device+0xe9/0x160
[    4.229261]  device_driver_attach+0x5d/0x70
[    4.229262]  __driver_attach+0x8f/0x150
[    4.229262]  ? device_driver_attach+0x70/0x70
[    4.229263]  bus_for_each_dev+0x7e/0xc0
[    4.229264]  driver_attach+0x1e/0x20
[    4.229265]  bus_add_driver+0x152/0x1f0
[    4.229266]  driver_register+0x74/0xd0
[    4.229268]  ? 0xffffffffc067b000
[    4.229269]  __wmi_driver_register+0x1a/0x20 [wmi]
[    4.229270]  clevo_wmi_driver_init+0x1c/0x1000 [clevo_wmi]
[    4.229272]  do_one_initcall+0x48/0x1d0
[    4.229274]  ? _cond_resched+0x19/0x30
[    4.229275]  ? kmem_cache_alloc_trace+0x37a/0x430
[    4.229276]  ? do_init_module+0x28/0x250
[    4.229277]  do_init_module+0x62/0x250
[    4.229278]  load_module+0x101c/0x1270
[    4.229280]  ? security_kernel_post_read_file+0x5c/0x70
[    4.229281]  ? security_kernel_post_read_file+0x5c/0x70
[    4.229282]  __do_sys_finit_module+0xc2/0x120
[    4.229283]  ? __do_sys_finit_module+0xc2/0x120
[    4.229284]  __x64_sys_finit_module+0x1a/0x20
[    4.229285]  do_syscall_64+0x38/0x90
[    4.229287]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[    4.229288] RIP: 0033:0x7f8aac62189d
[    4.229290] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d c3 f5 0c 00 f7 d8 64 89 01 48
[    4.229291] RSP: 002b:00007ffc23eb1d68 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    4.229292] RAX: ffffffffffffffda RBX: 000055b8c7e959f0 RCX: 00007f8aac62189d
[    4.229293] RDX: 0000000000000000 RSI: 00007f8aac4feded RDI: 0000000000000006
[    4.229293] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    4.229294] R10: 0000000000000006 R11: 0000000000000246 R12: 00007f8aac4feded
[    4.229294] R13: 0000000000000000 R14: 000055b8c7e70fc0 R15: 000055b8c7e959f0
[    4.229296] kobject_add_internal failed for tuxedo_keyboard with -EEXIST, don't try to register things with the same name in the same directory.

```

ftufo, 2021-02-25:
After upgrading to kernel 5.11, the problem seems to have disappeared.

tuxedoxt, 2021-02-26:
Hello, little late with the answer but I believe I saw it once as well, but not on 5.10. Usually does not happen and not consistently enough that I've been able to debug it.

Had a quick look and there should already be checks for existing objects which makes this a little bit weird that it sometimes happens.

Anyways I suspect it's not kernel version specific and a possible race condition during driver init.

Quick question, which device do you have? @ftufo

ftufo, 2021-02-26:
Hello,
I have an Aura15 running latest 5.11 kernel from Ubuntu mainline repository (see attached full dmesg)
[dmesg](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6048830/dmesg.log)
However the error does not seem to cause any issues other than the annoying message on boot.
Thanks.

tuxedoxt, 2021-02-26:
Hm ok interesting.

The Aura still has double interfaces, possibly when the wmi driver from your crash post gets loaded after the acpi driver... but it seems to have the same order in your kernel log from 5.11.

Anyways, thanks for the info.

ftufo, 2021-03-02:
Hello,
I attach the dmesg showing the error (just happened again), maybe it could be of help.

[err.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6067598/err.log)

ftufo, 2021-03-14:
It seems that both **clevo_wmi_probe()** of the _clevo_wmi_ module and  **tuxedo_keyboard_init()** ot the _tuxedo_keyboard_ driver, call **tuxedo_keyboard_init_driver()**. 
Sometimes, somehow, they call it concurrently, so a mutex is needed. Please see attached patch, I'm testing it on my Aura15.
[patch.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6136496/patch.txt)

tuxedoxt, 2021-03-15:
> It seems that both **clevo_wmi_probe()** of the _clevo_wmi_ module and **tuxedo_keyboard_init()** ot the _tuxedo_keyboard_ driver, call **tuxedo_keyboard_init_driver()**.
> Sometimes, somehow, they [...]

Yes they both call it for initialization since either or both can be present. This sound plausible.

Thanks for the patch, I will have a look.

tuxedoxt, 2021-03-22:
Included in release v3.0.4, as of today in deb/rpm repos also.

commit da49e9c2 (2023-04-17): Merge branch '75-stellaris-16-gen5-tdp-defs' into 'master'

Resolve "Stellaris 16 Gen5 TDP defs"

Closes #75

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!47
closed 2021-03-26 by ftufo