tuxedocomputers/tuxedo-keyboard #76: [Polaris] Touchpad not working after reboot

lldb3, 2021-02-14:
Hi, this is touchpad related, but not sure where to put it and there was previously an issue that [mentioned the same problem](https://github.com/tuxedocomputers/tuxedo-keyboard/issues/69#issue-798520769) so..

A small issue seems to occur every now and then. After a reboot, the touchpad would not work. Kernel message reads:
`i2c_hid i2c-UNIW0001:00: HID over i2c has not been provided an Int IRQ`

 It is the same as experienced [here](https://ask.fedoraproject.org/t/laptops-trackpad-not-always-recognized/9760/7) and the resolution is the same:
`rmmod i2c_hid && modprobe i2c_hid`. The link also has a workaround, but it's a dirty fix for an issue that shouldn't occur in the first place.

Not sure if this is fedora 33 only, would be nice to see if it's kernel, distro, or driver version dependent.

Anyway, I thought it would be good to have this issue here for anyone with the same problem.

ruffson, 2021-04-04:
Is there any update on this? This still happens and getting the trackpad to work after boot seems to be like a lottery. 
Seems like a race condition?

tuxedoxt, 2021-04-06:
Hello,

as far as I know we do not have a solution for this. In case it helps I believe I've seen it on opensuse with the recent AMD devices, but it does not happen on our standard ubuntu based installs.

> Is there any update on this? This still happens and getting the trackpad to work after boot seems to be like a lottery.
> Seems like a race condition?

I thought the same since reloading the i2c_hid driver seems to initialize it properly.

ruffson, 2021-04-06:
Okay thank you, hopefully this can be resolved in the future.

gagbo, 2021-04-27:
I think I have the same issue wuth a Pulse 15

tuxedoxt, 2021-04-28:
Very likely since it's the same CPU gen. Which kernel versions are you using btw? Basic support for these should be from 5.5 or 5.6 if I remember correctly.

gagbo, 2021-04-28:
I just installed Fedora 34 on it, so currently on `5.11.15-300.fc34.x86_64`, and I ran a few grub commands already I believe, so I'll put my cmdline : 
```
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-5.11.15-300.fc34.x86_64 root=UUID=[redacted] ro rootflags=subvol=root rd.luks.uuid=luks-[redacted] rhgb quiet
```
I had to boot 3 times to get touchpad working on the login screen, not sure what I missed

brunoais, 2021-05-06:
In my case, sending `{0x07, 0x03}` to `/dev/hidraw0` when the PC boots solves it.

nunocastromartins, 2021-05-06:
According to https://bugzilla.redhat.com/show_bug.cgi?id=1634832, where I believe the latest comments seem to describe the same bug, the fix should come to Fedora soon.

Spoil001, 2021-11-13:
I think I have the same issue with an infinitybook pro on pop. Kernel `5.13.0-7620-generic`.

My workaround was booting with a usb mouse attached, which seemed to fix the touchpad issue.

closed 2023-12-11 by tuxedo-bot