tuxedocomputers/tuxedo-keyboard #77: Arch Kernel 5.10.15 conf file not recognised at boot

DanCarzano, 2021-02-15:
Hi, I'm running this on a Metabox (Clevo) P95-HP and just updated the kernel to 5.10.15.
Since the last kernel update, my tuxedo_keyboard.conf file found in /etc/modprobe.d is not being recognised at boot. It always boots into the default white colors, however I had mine setup to default to Red. I've tried a few different fixes but nothing seems to work. That includes removing tuxedo-keyboard and its conf files and re-installing from both the AUR and compiling the modules myself. Any help on this would be greatly appreciated. Thanks

StevenSeifried, 2021-02-15:
I can confirm the issue.

AlexAnarcho, 2021-02-17:
I have the same issue running `5.10.15-1-MANJARO`. Running `sudo backlight dance` returns the following error 
[image]

Not too familiar with these modules though.

StevenSeifried, 2021-02-18:
@AlexAnarcho it looks you use "[TUXEDO Backlight Control](https://github.com/webketje/tuxedo-backlight-control)". This is third-party software which is not being developed by TUXEDO. 

After uninstalling "TUXEDO Backlight Control", the Fn keys should work

DanCarzano, 2021-02-23:
Just checking, has there been any updates to this issue yet? Thanks

astridlyre, 2021-03-11:
I am also experiencing this issue. Upon boot, it appears the `/etc/modprobe.d/tuxedo_keyboard.conf` file is not being read or used, as I also get the default white color. I am able to set the colors as usual, by echoing into `/sys/devices/platform/tuxedo_keyboard/color_*`.

I am running Arch Linux, kernel `5.11.4`. I just noticed this issue after the last tuxedo_keyboard update, which was on `Mon 08 Mar 2021` according to `pacman`. I'm running version `3.0.3-1`.

Trollwut, 2021-03-31:
Same for me, config file isn't being used.

Is there any update on this issue?

DanCarzano, 2021-04-01:
> Same for me, config file isn't being used.
> 
> Is there any update on this issue?
Nope

Trollwut, 2021-04-01:
Any idea on what is missing?

Maybe some other console-fu master drives by and can help with that problem.

Trollwut, 2021-04-05:
I found out something:

Upon booting, the file isn't being recognized and we're presented a white keyboard.

When I close my laptop lid, so that it falls asleep, upon awakening, the keyboard colors change to what is set in the file.

tuxedoxt, 2021-04-09:
Hello,

there were some fixes to module initiation included in `v3.0.4`. Please let me know if the problem is gone or still persists after installing this version and rebooting.

Trollwut, 2021-04-10:
Still the same problem for me with kernel 5.10

Going to test it with kernel 5.11 tonite or tomorrow, who knows.

StevenSeifried, 2021-04-10:
@Trollwut since kernel 5.11.12 I don't have this problem anymore

Trollwut, 2021-04-10:
Tested it with `5.11.10-1` and still not working...

Would you mind sharing your `/etc/modprobe.d/tuxedo_keyboard.conf` with me?

tuxedoxt, 2021-04-12:
> Tested it with `5.11.10-1` and still not working...
> 
> Would you mind sharing your `/etc/modprobe.d/tuxedo_keyboard.conf` with me?

If it works after wakeup (meaning values are set according to what is read from the file) the file format should be ok.

Could you check the version with `modinfo tuxedo-keyboard`?

DanCarzano, 2021-04-13:
Still not working for me either. Still getting the white keyboard at boot, fn keys are working however.
Version: 3.0.4
Kernel: 5.10.26-1

Trollwut, 2021-04-13:
> Could you check the version with `modinfo tuxedo-keyboard`?

Sure mate, here you go:

```
 modinfo tuxedo-keyboard
filename:       /lib/modules/5.11.10-1-MANJARO/kernel/lib/tuxedo_keyboard.ko.xz
softdep:        pre: tuxedo-cc-wmi
alias:          wmi:ABBC0F72-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F71-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F70-8EA1-11D1-00A0-C90629100000
version:        3.0.4
license:        GPL
description:    TUXEDO Computers keyboard & keyboard backlight Driver
author:         TUXEDO Computers GmbH <tux@tuxedocomputers.com>
srcversion:     4576AAC0AAC1CAA0E892D05
depends:        wmi,sparse-keymap
retpoline:      Y
name:           tuxedo_keyboard
vermagic:       5.11.10-1-MANJARO SMP preempt mod_unload
parm:           brightness:Set the Keyboard Brightness
parm:           color:Preset color for the keyboard backlight as string (string)
parm:           color_left:Color for the Left Region (uint)
parm:           color_center:Color for the Center Region (uint)
parm:           color_right:Color for the Right Region (uint)
parm:           color_extra:Color for the Extra Region (uint)
parm:           mode:Set the keyboard backlight blinking pattern
parm:           state:Set the State of the Keyboard TRUE = ON | FALSE = OFF (bool)
```

So the problem still persists: white keyboard (instead of my magenta) on boot. But if I sleep/wakeup, it suddenly changes to what I've set in the file.

DanCarzano, 2021-04-19:
Kernel upgraded to 5.10.30 and still not working as intended. Below is my output from modinfo tuxedo-keyboard

```
filename:       /lib/modules/5.10.30-1-MANJARO/kernel/lib/tuxedo_keyboard.ko.xz
softdep:        pre: tuxedo-cc-wmi
alias:          wmi:ABBC0F72-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F71-8EA1-11D1-00A0-C90629100000
alias:          wmi:ABBC0F70-8EA1-11D1-00A0-C90629100000
version:        3.0.4
license:        GPL
description:    TUXEDO Computers keyboard & keyboard backlight Driver
author:         TUXEDO Computers GmbH <tux@tuxedocomputers.com>
srcversion:     4576AAC0AAC1CAA0E892D05
depends:        wmi,sparse-keymap
retpoline:      Y
name:           tuxedo_keyboard
vermagic:       5.10.30-1-MANJARO SMP mod_unload 
parm:           brightness:Set the Keyboard Brightness
parm:           color:Preset color for the keyboard backlight as string (string)
parm:           color_left:Color for the Left Region (uint)
parm:           color_center:Color for the Center Region (uint)
parm:           color_right:Color for the Right Region (uint)
parm:           color_extra:Color for the Extra Region (uint)
parm:           mode:Set the keyboard backlight blinking pattern
parm:           state:Set the State of the Keyboard TRUE = ON | FALSE = OFF (bool)

```

tuxedoxt, 2021-04-19:
Thanks for the feedback.

tuxedoxt, 2021-04-19:
Cleaned up the init somewhat in between.

Not yet in a release but it's in the git repo (bea182ed6d2086698dcbb66c4f63bcaf1a5ac7c7) if someone is building manually please have a try. Would appreciate feedback.

Trollwut, 2021-04-19:
ayyyy, tested it and... it *kinda* works.

So I've set Magenta in the `/etc/modprobe.d/tuxedo_keyboard.conf` file.

While I rebooted with your mentioned commit, **it worked as intended**, yaaay!

Then I used `tuxedo-backlight-control` to set the color to Green.

After that, I closed my laptop lid to go into sleep mode.

I reopened my laptop and instead of either Magenta or Green, it turned to the standard Blue.

So yes, at boot it works now, but not after sleeping. (Sleep/awake though worked in the version before)

tuxedoxt, 2021-04-20:
> ayyyy, tested it and... it _kinda_ works.
>
> So I've set Magenta in the `/etc/modprobe.d/tuxedo_keyboard.conf` file.
>
> While I rebooted with your mentioned commit, **it worked as intended**, yaaay! [...]

Weird, for my test case I don't have any issues. The resume uses the same hooks as usual. There should not be any difference with this part.

- Which laptop model?
- Which sleep mode?

Trollwut, 2021-04-20:
> Which laptop model?

Base model should be a `CLEVO P775TM1-G` if I'm not mistaken.

> Which sleep mode?

I use the trigger via KDE on laptop lid close:

[image]

So I *think* that should be `suspend` instead of hibernate. Is there any way to confirm this?

TBH I don't use suspend that often, so that new bug doesn't affect me too much. Just pointed that out for other users who might use it more often. :)

tuxedoxt, 2021-04-20:
> Base model should be a `CLEVO P775TM1-G` if I'm not mistaken.
>
>
> I use the trigger via KDE on laptop lid close:
>
> [image]
>
> So I _think_ that should be `suspend` instead of hibernate. Is there any way [...]

You can look for a line in dmesg like `PM: suspend entry ()`

Trollwut, 2021-04-20:
Hope this suffices for you :)

```
 sudo dmesg | rg suspend                                                                       1796ms
[32933.584733] PM: suspend entry (deep)
[32935.011923] printk: Suspending console(s) (use no_console_suspend to debug)
[32938.315990] PM: suspend exit
[32938.573830] audit: type=1130 audit(1618938174.403:318): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[32938.573833] audit: type=1131 audit(1618938174.403:319): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-suspend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```

Trollwut, 2021-04-20:
Just wanted to say that today this problem did NOT accure, in like, it worked as intended (boot + sleep/awake).

Give me some more days to investigate, if this was a simple mistake from my side or if I can reproduce this error.

tuxedoxt, 2021-04-26:
Included the latest fix in the `v3.0.5` release.

Feedback is still very welcome.

Trollwut, 2021-04-29:
Tested a few days: Boot loads the config and also lid close/open does it.

Seems like it's been resolved! :)

DanCarzano, 2021-05-10:
Sorry it took me a while to get back to you guys about this. But I can happily confirm the issue has been resolved! Thankyou :)

commit bef6df9b (2023-05-15): Merge branch 'add_ibp_gen8_leds_support' into 'master'

Add ibp gen8 leds support

Closes #77

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!50