tuxedocomputers/tuxedo-keyboard #78: PC goes crazy after a while

B-Derouet, 2021-02-24:
After a long time of use, my PC goes completely crazy. The keyboard is completely stuck on the last keys used (so it's impossible to use it), the sound bug (repeated passages several times), I'm randomly disconnected from my internet box and my PC doesn't want to shut down at all (having an internal battery, I have to let it discharge completely). The only way I've found to keep using my PC when this happens is to disconnect/reconnect to my user account and activate the visual keyboard, my keyboard being indefinitely blocked, reconnection at least allows me to have the last used keys disabled. I had to completely uninstall tuxedo-keyboard so I don't have this problem anymore.

Bug always present since v2, even on the last version (3.0.2). Please find in attachment an extract of the report generated by the `dmesg` command.
[log.txt](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6036899/log.txt)

tuxedoxt, 2021-02-24:
Hello,

which device do you have?

The problem is pretty characteristic and in short needs an update of the EC to fix.

B-Derouet, 2021-02-24:
I have a Clevo PC50DD2 and I just realized that my BIOS version wasn't the same as my EC version... I'll update it and see if the problem is solved :)

tuxedoxt, 2021-02-25:
BIOS and EC are not necessarily on the same version number.

Which EC version do you currently have?

B-Derouet, 2021-02-25:
Ok, this is what I saw last night, the versions can be different. But after checking with the manufacturer, no update is available for my EC, so I'm forced to uninstall tuxedo-keyboard and tuxedo-control-center and wait...

InsydeH2O BIOS v1.07.09
EC v1.07.02

tuxedoxt, 2021-02-25:
Ah right yeah it's not one of our versions. That makes it more complicated.  You might need to ask your manufacturer for an update.

demib72, 2021-03-28:
Having a similiar issue with my Clevo NH57ADS, where the keys get stuck on the last key hit and I can't even use the power button to force a restart. Bios version is Insyde Corp 1.07.07. Not entirely sure how to check the EC version, exhausted my google fu powers.

Log: https://zerobin.net/?12410ee85b61cbff#cOCHHv2gSK4wykYu7UsFmyGQDdY1v+7AnCVy8FTZLpE=

tuxedoxt, 2021-03-29:
Yep, this is the same issue. EC update is needed. You can check your whole EC version in the "bios" menu/info screen. (Boot with F2 or from OS reboot with `systemctl reboot --firmware-setup`)

Green-Sky, 2021-04-19:
I had the same problem with my XA1510, so i flashed the EC 1.07.04TR1 -> 1.07.04TR3.
Hope that fixes this.

srd424, 2021-10-31:
Hi, I've got just bought a Clevo NL51CU (from another reseller unfortunately!) and have experienced the keyboard locking up with nothing in the logs, with or without tccd running. I appreciate you don't support this laptop, but just wondered if you knew if it suffered from the same/similar EC bug? As far as I can tell it is running on latest versions, BIOS is 1.07.14LWL2, EC is 1.07.09 ..

commit 77ca765b (2023-05-15): Merge branch '78-ibp-gen8-tdp-defs' into 'master'

Resolve "IBP Gen8: TDP defs"

Closes #78

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!52
closed 2023-05-15 by tuxedo-computers