tuxedocomputers/tuxedo-keyboard #80: personal backlight colors not loaded anymore

puppenklo, 2021-03-04:
Since a couple of days the backlight colors behave strange. First thing that happened is, that only the left side of the keyboard had the default white color, while the middle and right parts still had their custom colors after starting my machine. When I rebooted the system the white color took over the middle part, too and at the next boot all the backlight was white colored. I changed already every single _tuxedo_keyboard.conf_ that could be found to my custom colors, but still every part is white after rebooting or starting the machine.
When I tried to restart the module via
`sudo rmmod tuxedo_keyboard`
`sudo modprobe tuxedo_keyboard`
it gave back this error:

> rmmod: ERROR: Module tuxedo_keyboard is in use by: clevo_acpi tuxedo_io clevo_wmi

First I was irritated because it was easily possible in the past but then I tried to remove those in different orders and reload them reversely ordered with, but without success.
What did the trick was reloading _clevo_acpi_ after reloading _tuxedo_keyboard_.
If the other modules have any influence on the process is out of my knowledge, yet.

So I guess the problem is that something is wrong since some update in the past days and now the modules load in a different order and there are now dependencies that have not been there before. In the past it was possible to unload and load the _tuxedo_keyboard_ module without any problems.

Does anyone have the same problem, or is it just me?

Greetings,
puppenklo

tuxedoxt, 2021-03-04:
Hello,

regarding the backlight this seems related to #77. Thanks for the info, this is under investigation.

Regarding the module structure, this is by design. Pushed by new variations of hardware interfaces in new devices this part was modularized.

As you discovered reloading is still possible but you have to unload the interface drivers first. As a reference, here's a link to the install script which does just that.

https://github.com/tuxedocomputers/tuxedo-keyboard/blob/4296f75372213c37a022f6ccff4340ea7063a223/deb/module-name/DEBIAN/postinst#L59-L68

puppenklo, 2021-03-09:
Hi,

don't know if the last update of _tuxedo-keyboard_ changed anything related to the problem, but it didn't occur since then. On the other hand it was never a permanently existing problem. After the first occurrence it worked fine again for some days and then it came back later. I will report here at the end of March, if it came back or not.

tuxedoxt, 2021-03-09:
There was no intentional fix in the last update.

puppenklo, 2021-03-30:
After not returning for this month the problem occured again a few days ago. Actually just after starting my notebook there was an update of the _tuxedo-keyboard_ module. After this update the colors changed back to my prefered setting. No problems since then.

tuxedoxt, 2021-04-09:
> After not returning for this month the problem occured again a few days ago. Actually just after starting my notebook there was an update of the _tuxedo-keyboard_ module. After this update the colors changed back to my prefered setting. No problems since then.

Great, some thing related to module initiation were fixed. Please let me know if the problem returns.

closed 2021-04-09 by tuxedoxt

9WinterMute6, 2021-07-16:
Hi, here is a Polaris 15 notebook which has this issue over and over again. Code above that removes and reloads the modules leads to a short flash of the backlight with the configured colour and then nothing. Also while the backlight does not work, the keyboard layout is slightly different, too: 'INS SCR' is DEL, DEL is move to the end of the line and 'Dot' is DEL.