tuxedocomputers/tuxedo-keyboard #81: Potential multithread issue in clevo_keyboard.h

ftufo, 2021-03-18:
Hello,
while trying to understand why I get repeated messages from tuxedo_keyboard at boot , I found a potential multithreading issue.
[    4.652059] kernel: tuxedo_keyboard: set_mode on CUSTOM
[    4.658227] kernel: tuxedo_keyboard: set_mode on CUSTOM
[    4.684015] kernel: tuxedo_keyboard: Set brightness on 75
[    4.697351] kernel: tuxedo_keyboard: Set brightness on 75
[    4.701569] kernel: tuxedo_keyboard: Set keyboard enabled to: 1
[    4.708329] kernel: tuxedo_keyboard: Set keyboard enabled to: 1

The **clevo_keyboard_add_interface()** function is called by both the _clevo_wmi_ and the _clevo_acpi_ drivers, as far as I understand the Aura15 has both the interfaces, so it gets called twice (that should also be the reason of the doubled messages at boot), but the acpi interfaces should always be used when present.

The problem is in the lines:
active_clevo_interface = clevo_interfaces.wmi;
...
active_clevo_interface = clevo_interfaces.acpi;

Since the active_clevo_interface variable is global and two different drivers could modify it at the same time, i think a mutex should be used to be safe. From the kernel timings above, it seems that the two drivers are loaded concurrently (but don't know what could happen when a wmi interface is used when also the acpi one is present, maybe nothing bad :-)).
Please find attached a proposed patch.

Thanks.
[patch.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6163414/patch.log)

tuxedoxt, 2021-03-18:
Hello,

as far as observed both interfaces work the same. Can partly be verified in the ACPI implementation and is basically a "copy paste" interface.

Nothing bad should happen. However both should not be called. I have not seen this one yet. A modified version of your patch for the other issue is committed but not yet released. I'll verify that this also includes the active interface before release to be on the safe side.

tuxedoxt, 2021-03-19:
Explicitly added another mutex for interface modifying operations 67fb677d886c7808fbca56eae9136b19381ea020

tuxedoxt, 2021-03-22:
Included in release v3.0.4, as of today in deb/rpm repos also.

closed 2021-03-26 by ftufo