tuxedocomputers/tuxedo-keyboard #82: FN Backlight Keys not working

shiftzwei, 2021-03-23:
After installing tuxedo-keyboard fn-F6 and fn-F7 Keys no longer working. All other fn keys working fine. 

    OS: Manjaro Linux x86_64
    Host: XMG FUSION 15 (XFU15L19) Late 2019
    Kernel: 5.11.6-1-MANJARO

Currently installed:

    extra/tuxedo-control-center 1.0.11-1 [Installiert]
    extra/tuxedo-keyboard 3.0.2-1 [Installiert]

I tried tuxedo-keyboard latest release 3.0.4 with same error.

    Mär 23 17:04:46 laptop gsd-power[1787]: gsd_power_backlight_abs_to_percentage: assertion 'max > min' failed
    Mär 23 17:04:46 laptop kernel: atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
    Mär 23 17:04:46 laptop kernel: atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 2f with keysym 37 (keycode 10).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 73 with keysym 73 (keycode 27).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 39 with keysym 39 (keycode 12).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 36 with keysym 36 (keycode f).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 37 with keysym 2f (keycode 10).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym ff53 with keysym ff53 (keycode 72).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 35 with keysym 35 (keycode e).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 73 with keysym 73 (keycode 27).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym ff52 with keysym ff52 (keycode 6f).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 6f with keysym 6f (keycode 20).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym ff51 with keysym ff51 (keycode 71).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 61 with keysym 61 (keycode 26).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym ff54 with keysym ff54 (keycode 74).
    Mär 23 17:05:12 laptop gnome-shell[1593]: Window manager warning: Overwriting existing binding of keysym 38 with keysym 38 (keycode 11).
    Mär 23 17:05:21 laptop kernel: atkbd serio0: Unknown key pressed (translated set 2, code 0xf8 on isa0060/serio0).
    Mär 23 17:05:21 laptop kernel: atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
    Mär 23 17:05:23 laptop kernel: atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
    Mär 23 17:05:23 laptop kernel: atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.

Thank you

tuxedoxt, 2021-03-23:
Hello,

for fusion you'll need the (still separate) `tuxedo-keyboard-ite` module package as well.

tuxedoxt, 2021-03-23:
Also I do not see any errors related to `tuxedo-keyboard` or its modules specifically.

shiftzwei, 2021-03-23:
Hi, wow thank you. Installing tuxedo-keyboard-ite did it.

closed 2021-03-23 by shiftzwei

akssieg, 2021-09-12:
I have XMG NEO 15.

OS: Ubuntu 20.04.1
Host: XMG NEO 15 (EARLY 2019)
Kernel: 5.4.0-84-generic

Installation steps followed from: https://www.tuxedocomputers.com/en/Infos/Help-Support/Help-for-my-device/XMG-FUSION-series/XMG-FUSION-15/Intel-QC7-Ubuntu-20-04-installation.tuxedo

and installed 

```
5.1 tuxedo-keyboard
5.2 tuxedo-keyboard-ite
5.3 tuxedo-tomte
5.4 tuxedo-control-center
```
Backlight is not working even though I changed configs inside "sudo nano /etc/modprobe.d/tuxedo_keyboard.conf". Any help would be appreciated.