tuxedocomputers/tuxedo-keyboard #83: RGB light stays blue and cannot be turned off

CFDaniel1893, 2021-04-16:
Hello everybody,

I am using a XMG Core 15 with Ubuntu 20.04. and the Tuxedo Control Center.
Earlier I was able to turn on/off the RGB light with Fn+Space. 
After installation the RGB light stays blue and I cannot turn it off or change the mode or color of the light.
Do you know what the problem could be? 
I appreciate any kind of help. 
If you need additional information please ask.

Thank you very much
Daniel

corus87, 2021-04-18:
@CFDaniel1893 
Tuxedo does not support XMG Laptops, after you installed the Tuxedo control center, you also installed the tuxedo-keyboard kernel module, now your Ubuntu uses the tuxedo-keyboard driver and since your XMG is not supported, it won't work anymore.
However, I also use a core 15 with Linux, check out my [keyboard script](https://github.com/corus87/tuxedo-xmg-keyboard-backlight-control-script) where I describe how to support the core 15 with the tuxedo-keyboard module.

CFDaniel1893, 2021-04-18:
Many thanks for your help @corus87 

I just tested your script but I got following error:
tee: /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/color_string: No such file or directory
tee: /sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/brightness: No such file or directory

Ma you please also share the files I need additionally or tell me what I have to write into these files?

Many thanks!

corus87, 2021-04-18:
Did you compile tuxedo-keyboard and add your device to it as mentioned in the readme? 

But besides that, better open a issue [here](https://github.com/corus87/tuxedo-xmg-keyboard-backlight-control-script) since this is not a tuxedo related issue.

CFDaniel1893, 2021-04-18:
Problem is solved here:
https://github.com/corus87/tuxedo-xmg-keyboard-backlight-control-script/issues/1#issue-860679274

closed 2021-04-18 by CFDaniel1893