tuxedocomputers/tuxedo-keyboard #86: Polaris Gen1: Control power mode LED and what it does

caravelaPortuguesa, 2021-04-26:
Hello,
I'm looking for a way to, if possible, control the power mode lights.
If not possible, just being able to read the value is enough.
Then I want to use that value in user mode to change the tcc profile without the need for a password. <- This one I know how.

I already know _tuxedo-keyboard_ locks the power mode value to the bottom LED ON.
It also appears like _tuxedo-keyboard_ is opened through a keyboard shortcut that is activated here: https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/uniwill_keyboard.h#L253
(I figured that out because  I saw this a few days ago made by the user brunoais: https://github.com/tuxedocomputers/tuxedo-control-center/issues/101)

I already asked support e-mail but I got this as answers:

> Our main developer asked if you could post your question on Github, since that makes it easier for him too.
> The short version is:
> - The LED **cannot** be controlled! Why the EC Controller or the Control Center under Windows behave differently, my colleague will be able to answer that better.
> - It should be possible to remap the button, so that instead of the Tuxedo Control Center, something else can be opened.

Do you confirm the LED cannot be controlled and that it's status cannot be read?
Can the button be remapped to open a different program instead of just pressing some keys?

tuxedoxt, 2021-04-26:
Yes, the support answer is correct. These LEDs can unfortunately, as far as we know, not be controlled separately. This was at one point actually also confirmed by the ODM. Still, I tried to look for it anyways ^^ however, so far without success. If you find a way I'd be very interested.

The LEDs seem to change depending on which ODM power profile is chosen. This can be read and possibly mapped (see ec ram address `0x0751` if you tumble around in the driver). For the moment we only use one profile, more or less, to allow fan control.

The key event from the button is, as you discovered, caught in the driver and thrown as a key code combo. This is currently not configurable. You could possibly catch this combo and throw a new one of your choice or modify the driver.

caravelaPortuguesa, 2021-04-26:
I'll see if I can figure out. From what it seems, support relayed to me it cannot be done. You say it might be doable. You mention ODM (original device manufacturer(?)) says it cannot be done. Did I get that right?
However, that can't be right. How does the software in windows know which power profile to be in? Even if just to make that icon appear in the screen? I wonder if the ODM is hiding something.
What does `tuxedo-keyboard` do to turn it off and fixed into 1 LED on, btw?

tuxedoxt, 2021-04-26:
Some things can be done, some not, let's keep things separate.

- The LEDs themselves cannot be directly controlled without affecting other things.
- The ODM power profiles can be set and read. This indirectly changes the LEDs (as far as we know hardcoded through EC)

> What does tuxedo-keyboard do to turn the LED changing off and fixed at 1 LED on, btw?

It sets (at least) one bit connected to the ODM power profiles to allow us a way to manually control the fans. This indirectly affects the LEDs.

We'd rather connect the LEDs to our TCC profiles for example, but again, as far as we know and confirmed by the ODM this is not possible.

caravelaPortuguesa, 2021-04-26:
I see... Thank you.

caravelaPortuguesa, 2021-04-26:
1. Do the bits have to be set in such way that 1 LED is ON so the fans can be controlled and override ODM's power profile?
2. So I have where to start with, what are the bits that control the LED (which also activate ODM power profiles)? (not knowing is OK but I'm not sure if that's the case)

brunoais, 2021-04-26:
Hello! I noticed this discussion and I have something to add.
I've been trying to get tuxedo_keyboard to work with different power modes and I have been trying to reverse-engineer how the LED for the power mode work and I think I got it.
Power mode is in the address `0x0751` like @tuxedoxt mentions. It's actually 2 bits at `0x80` and `0x10`.
`0x80` is LED off
`0x10` is both LED on
^ Set both at 0 and you get 1 LED on. That's the reason why the button is at 1 LED on when tuxedo_keyboard starts and in some ioctl calls.
The other bits control other features I didn't focus on. One of them being the manual mode. I don't know which one is the manual mode yet.

Here's the catch. Unlike what @tuxedoxt mentioned. It appears like the ODM lied. If the mode is changed by pressing the button, the fan profile changes to a predefined one **until** an order is done otherwise. Because `tccd` keeps giving orders otherwise, it doesn't matter at all. If the fans or the CPU profile changes, it's completely unnoticeable. I made a `while True: pass` in python in multiprocess to test under load. The power mode LED changes seem to do nothing.

You can find my WIP here: https://github.com/brunoais/tuxedo-keyboard/tree/power_mode_works
So far, there's a sysfs file at: `/sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/power_mode`.
You can read the current power mode and write the power mode you want, using it. You can make it blink using that:

`while true; do echo "2" > power_mode; sleep 0.5; echo "0" > power_mode; sleep 0.5; done;`

Or an interesting ramp up:
`while true; do echo "0" > power_mode; sleep 0.2; echo "1" > power_mode; sleep 0.5; echo "2" > power_mode; sleep 1; echo "1" > power_mode; sleep 1; done;`
(I show you only to exemplify the full control)

I just finished the ioctl changes.

The only thing I still want to make is a file stream to write power mode changes, so programs can readblock on them and receive 
I was thinking to make it on `/sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/power_mode_notify`. Afterwards, create the PR.

Ideally, this would allow hooking 3-5 modes to tuxedo-control-center profiles. 3-5 because if you use blinking, you can squeeze extra 2. However, officially, I think 3 should be the choice and leave the extra 2 to more expert users.
If I were to decide on that, it would be a driver parameter. However, this would require proper locks and threads. It would definitely be something for later.

The thing that bothers me now the most is the way tccd changes profiles. It should be a pipe "file" instead of the need to restart it every time. That restart process causes the fans to ramp up unnecessarily.

caravelaPortuguesa, 2021-04-28:
Tuxedo, is brunoais claim true? Were you able to confirm it?

caravelaPortuguesa, 2021-05-06:
Tuxedo, any updates?

brunoais, 2021-05-06:
Try mentioning tuxedo like so: @tuxedoxt

brunoais, 2021-05-10:
Here I show a video

https://user-images.githubusercontent.com/639467/117719354-34633780-b1d5-11eb-8389-e6886e9734ae.mp4

https://user-images.githubusercontent.com/639467/117719368-3927eb80-b1d5-11eb-9bea-5f91c1e105a5.mp4

https://user-images.githubusercontent.com/639467/117719379-3e853600-b1d5-11eb-9a4e-d64e4c140d14.mp4

Here's the full video before being cut (remove the `.log` extension from the files):

[performance not affected.mp4.7z.001.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454789/performance.not.affected.mp4.7z.001.log)
[performance not affected.mp4.7z.002.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454790/performance.not.affected.mp4.7z.002.log)
[performance not affected.mp4.7z.003.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454791/performance.not.affected.mp4.7z.003.log)

brunoais, 2021-05-10:
> However feel free to prove it wrong,

Here I show a video where the power profile changes but TCC still has full control over the fans (had to be cut due to file size constraints):

https://user-images.githubusercontent.com/639467/117719354-34633780-b1d5-11eb-8389-e6886e9734ae.mp4
https://user-images.githubusercontent.com/639467/117719368-3927eb80-b1d5-11eb-9bea-5f91c1e105a5.mp4
https://user-images.githubusercontent.com/639467/117719379-3e853600-b1d5-11eb-9a4e-d64e4c140d14.mp4

Here's the full video before being cut (remove the `.log` extension from the files):

[performance not affected.mp4.7z.001.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454789/performance.not.affected.mp4.7z.001.log)
[performance not affected.mp4.7z.002.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454790/performance.not.affected.mp4.7z.002.log)
[performance not affected.mp4.7z.003.log](https://github.com/tuxedocomputers/tuxedo-keyboard/files/6454791/performance.not.affected.mp4.7z.003.log)

> ODM power profile choice to the TCC profiles as long as we can keep our fan control.

Is what I shown enough to prove that? From what it seemed to me, TCC had full control over the fans during the experiment. Power mode didn't do anything about how they were turning. It was all TCC. Feel free to explain me wrong ;).

I forgot to check with zenmonitor so I went to do it in a new recording.
https://user-images.githubusercontent.com/639467/117723033-c9682f80-b1d9-11eb-82a8-f4f76f44fc92.mp4
In manual mode, it appears like the LED gets changed but nothing happens to the PC besides the power mode LED (you can see in the current setting in the "Power mode" terminal window).

>  From memory there are three modes 0x00, 0x10, 0xa0

In Polaris 17 I have (nvidia and AMD), it's actually `0x00, 0x10, 0x90` but either 0x90 and 0x80 give the same result in all aspects I could test by reverse-engineering. What does the 0x20 bit do? I really wish I had the manual at this point but I can't find it anywhere.

> most lightweight devices only have two

Oh! Any more information so I can code that down? For now, I have the 3 power modes hardcoded because I don't know how to detect them.
You can find the code here: https://github.com/brunoais/tuxedo-keyboard/tree/power_mode_works
It is still not clean, btw.

> Some regions might make more persistent and surprising changes ^^

Agreed. I found a bit to make fans go 100%. Makes me want to assign that feature to `Fn`+ `F1` just like my previous PC. Took me quite some tries to single out the two bits that control the power mode. I actually didn't expect that none set is not the lowest power mode and is the medium instead.
I still want that manual....

brunoais, 2021-05-12:
Meanwhile, I made a video showcasing different settings on TCC and me changing power modes. During that process, nothing visibly happens when power mode changes. Changes only happen when the profile is changed.

https://user-images.githubusercontent.com/639467/117930459-eabe3e00-b2f5-11eb-834d-78821609b7c1.mp4

My suspicion is that those two bits make changes to the choice of fan speed and CPU frequency but TCC overrides them by setting its own. I only have 1 machine (and only have money for 1 machine) so I can't execute these tests on other machines with my version of the driver for comparison.

caravelaPortuguesa, 2021-05-12:
Brunoais, 
What should I do to install your version instead?

brunoais, 2021-05-12:
@caravelaPortuguesa 
Don't. I really recommend you not to. Neither have I installed the driver in case it would cause a problem. When I boot my PC the driver loaded is the stable version from this repository.
**I'm still not done testing and it can break at any time requiring you to reboot your PC to fix it.**
### Do __NOT__ do it
**unless you are willing to deal with such potential issues and having `dmesg` in hand to debug or fix what happened.**

Having said that, assuming you understand the risks, the compilation procedure is the same as in this repository. Just run `make` and then you can run the convenience script `reloadModules.sh` to reload the driver.

caravelaPortuguesa, 2021-09-13:
Is it now the time to install your version?

caravelaPortuguesa, 2022-11-15:
@brunoais It's been one year. Is this ready yet?

brunoais, 2022-11-27:
@caravelaPortuguesa No. It's not merged, so it's not ready.