tuxedocomputers/tuxedo-keyboard #88: Polaris 17 keyboard back light is not controllable with tuxedo-keyboard

RecNes, 2021-05-04:
Actually, I have a same computer with different branding. 
(https://www.monsternotebook.com.tr/abra/monster-abra-a7-v11-2-2/)
The brand I bought has no Linux support. So I decided to figure out what kind of device I have an install necessary modules to control special options like CPU speed, keyboard back light, fan speed etc.
I did some research and I find the Polaris 17. shape of case of my machine and Polaris 17 is exactly same. 

I did installation steps in README.md but no luck for my device.
Is there a way to confirm the keyboard on my device is the same as Polaris 17?
If I can whether confirm or not my machine have same keyboard, that can give me a clue for tuxedo module could do the trick or not.

brunoais, 2021-05-05:
I'd bet the firmware is different, so the reporting and the OS detection makes it incompatible with `tuxedo-keyboard`

RecNes, 2021-05-11:
> I'd bet the firmware is different, so the reporting and the OS detection makes it incompatible with `tuxedo-keyboard`

I'm wondering which firmware you are talking about, keyboard or bios?

brunoais, 2021-05-12:
@RecNes EC (Embedded controller)

kgensou, 2021-05-13:
I have a similar issue with the XMG Fusion 15. Fn+F6/F7 doesn't control the backlight until the tuxedo-io module is unloaded. Backlight control works fine during boot/BIOS until it is loaded.

brunoais, 2021-05-14:
I went to check the code. 
https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/uniwill_keyboard.h#L456-L465

If I got it right, nothing happens with those PC because `tuxedo-keyboard` doesn't recognize the BIOS version you are using. Even if it is the same-looking as the Tuxedo PCs, they show themselves to the driver as being different, so the driver can't recognize them.

The code is written with name detection like this because the hardware manufacturers didn't document (maybe even implement) means to read what model of keyboard is installed. So they have to resort to feature detection.

However, the keyboard keys default shortcuts stop working  because [manual mode is turned on](https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/tuxedo_io/tongfang_wmi.h#L295).

That's how your `Fn` controls stop working. To fix this, you will need to propose the driver changes required to include your PC. I don't think Tuxedo will spend time with PCs that are not Tuxedo.

whataboutthat, 2021-05-14:
thanks brunois for pointing out the line.
what bothers me is that the other controls still work. I can still use the FN key + F1..F8 to adjust the volume, the screen brightness etc. Only the combination of FN+space doesn't work.

onurbbr, 2021-09-18:
@RecNes I'm also a Monster user. I am using Tulpar T7 V20.2 (Same generation and same spec clone of Polaris 17 Gen1)(RTX 2060 + i7-10750H). I solved it in a slightly different way. I found and extracted the bios reset file provided by XMG company and changed a few things necessary for the keyboard. My keyboard works in a single color and any color I want, and I solved this problem by turning off RGB support. I don't bother much for the instant color change. Maybe this will change in the future.

onurbbr, 2021-09-25:
> I went to check the code.
> https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/uniwill_keyboard.h#L456-L465
> 
> If I got it right, nothing happens with those PC because `tuxedo-keyboard` doesn't recognize the BIOS version you are using. Even if it is the same-looking as the Tuxedo PCs, they show themselves to the driver as being different, so the driver can't recognize them.
> 
> The code is written with name detection like this because the hardware manufacturers didn't document (maybe even implement) means to read what model of keyboard is installed. So they can't resort to feature detection.
> 
> However, the keyboard keys default shortcuts stop working because [manual mode is turned on](https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/tuxedo_io/tongfang_wmi.h#L295).
> 
> That's how your `Fn` controls stop working. To fix this, you will need to propose the driver changes required to include your PC. I don't think Tuxedo will spend time with PCs that are not Tuxedo.
> 
> If you are feeling adventurous
> 
> ### **AND BE CAREFUL!!! Do NOT install a driver you compiled yourself without WEEKS of testing. Drivers can do ANYTHING, INCLUDING making your PC UNBOOTABLE**
> you can do the changes yourself to remove the https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/src/uniwill_keyboard.h#L456-L465 detection, make it always true and compile the code yourself.

There is a problem with the name of the DMI board. Adding your own DMI board name solves the problem. This is how I solved it. I am using Monster Tulpar T7 V20.2. You can find the board name by using the command "cat /sys/devices/virtual/dmi/id/board_name" and you can run this step on the device (applies to one-way lighting).

RecNes, 2021-09-27:
@onurbbr Thank you for your comments. I'll be try your solutions. If it works, I'll post a comment here.

onurbbr, 2021-09-28:
@RecNes If the problem is not resolved, you can reach me via Telegram by specifying your github username. Telegram: @moonaru

tustunkok, 2022-06-19:
> There is a problem with the name of the DMI board. Adding your own DMI board name solves the problem. This is how I solved it. I am using Monster Tulpar T7 V20.2. You can find the board name by [...]

I can confirm that this method definitely works with Monster Abra A5 V17.1 as well. Thank you so much.

closed 2023-12-11 by tuxedo-bot