tuxedocomputers/tuxedo-keyboard #90: Pulse 15 Gen1: Can't control backlight anymore

SkaveRat, 2021-05-13:
Not entirely sure if this is the same problem as reported in #88.

I had to reboot my laptop today and I can't control the backlight with Fn+Space anymore.

I don't know what update broke it, as my system had a couple weeks of uptime.

The light works during the bios boot phase, but afterwards it's just dead.

BIOS settings say that it's enabled.

tuxedoxt, 2021-05-14:
Hello,

no, this is unrelated to #88.

Please reboot to bios settings (F2 while booting or `systemctl reboot --firmware-setup` in linux) and check BIOS and EC version.

SkaveRat, 2021-05-19:
versions are `N.1.07.A02` and `1.07.30.00`

I've noticed that it works again after suspending the machine and waking it up

fabrixxm, 2021-05-20:
Same issue on my Pulse 14 Gen1.
BIOS Version N.1.07.A02
EC Version 1.07.30.00

suspend and resume make it works

rucub100, 2021-06-27:
I solved the problem on my Pulse 15 Gen1.
BIOS Version N.1.07.A03
EC Version 1.07.30.00

the trick is to make sure you disable backlights on start and enter the bios, here I switched brightness settings from  level1 to disabled, so that what I see and what I (no backlight) get is consistent; finally save settings and restart. Works so far, however I guess one could break is again in bios if you make is inconsistent. So it feels more like a workaround to me.

tuxedoxt, 2021-06-28:
> I solved the problem on my Pulse 15 Gen1.
> BIOS Version N.1.07.A03
> EC Version 1.07.30.00
>
> the trick is to make sure you disable backlights on start and enter the bios, here I switched brightness [...]

Hello, did you happen to find a way to trigger the issue (more or less reproducable) as well?

rucub100, 2021-06-28:
> Hello, did you happen to find a way to trigger the issue (more or less reproducable) as well?

Hi @tuxedoxt, I tried to change settings in BIOS and with FN+Space (at the same time) to make it inconsistent again. However I am not able to reproduce the issue. Either the fix is permanent or the problem is not the inverse of the solution I described above.

Note: I encountered the problem right after I finished my new OS installation (archlinux), and before that I did the EC and BIOS updates.

thepasto, 2021-07-03:
Thanks @rucub100 it seems to solve the problem! i [reported](https://github.com/tuxedocomputers/tuxedo-keyboard/issues/68#issuecomment-797328924) the problem some month ago. I'm happy you found a solution. Thanks again

amithiel, 2021-08-21:
Hi!
Brand new pulse 15 here. Since i own this laptop (few days), i have yet to find a linux distro that works for more than 24h.
As i type this, i have no lights on the keyboard. Last night they were working, i woke up...rebooted the machine and bye bye.They turn on at the grub menu, as soon as the OS boots, they turn off forever.
I'm losing my patience honestly.
I have installed tuxedo-keyboard, tuxedo-control-center, and also the tuxedo-touchpad-switch (which does nothing here on linuxmint, however it works  on archlinux).

thepasto, 2021-08-21:
fastest fix about broken keyboard backlight, is to perform a suspend, when you'll wake up your laptop, the backlight is working as well. it's not a real fix and it's a bit boring but i didn't find any better.

amithiel, 2021-08-21:
> fastest fix about broken keyboard backlight, is to perform a suspend, when you'll wake up your laptop, the backlight is working as well. it's not a real fix and it's a bit boring but i didn't find any better.

Yeah, well, kinda wish i knew that before buying. I guess i din't research enough

amithiel, 2021-08-22:
What did i tell you?  Can't handle 1 day without breaking. Yesterday i rebooted several times, the lights were always working like a charm. Everything went fine, i used the laptop all day.
This morning, resumed from sleep...all fine. 
Just made a reboot now, and BAM, lights gone. I can confirm however that closing the lid (suspending) and resuming again, bring them ON. Which i guess is something part of my life now

thepasto, 2021-08-22:
It would help to have an "official" answer about this issue, i think we can't do more by ourselves

amithiel, 2021-08-22:
I'm not quite sure what logs would be helpful here, to help them fix this in a future EC patch or whatever. But i'll help in any way possible.
I looked at journalctl after the reboot when the lights went off, but i'm not that expert. Didn't find something weird that i can share.
However, just let me know what would be useful, so the next time it happens i can capture the right logs.
At least i'm relieved its not related to kernel updates. For now i'm sticking to linuxmint and the tuxedo repos and this i figured everything about this issue

tuxedoxt, 2021-08-23:
Hello,

thanks for the info. The issue has been reported on and off. For awhile it was thought to be solved with the latest firmware. So far no consistent way of reproducing it is known. We are very interested to get it solved. As much info as possible about the incidents helps.

From the latest comments it seems like it breaks and "unbreaks" at boot and wakeup. Also possibly breaks right after flashing.

Other things that would be interesting to note when it happens.
- Battery mode or "AC mode"?
- Linux or Windows?
- Firmware versions BIOS/EC
- `tuxedo-keyboard` installed? version?
- Other devices connected?

amithiel, 2021-08-23:
Hi!

managed to trigger it, right now consistently. First, answering your questions:

- Battery mode / ac mode (doesn't matter)
- Linux (the distro doesn't really matter)
- Tuxedo keyboard installed: V3.0.8
- No other devices connected

To trigger:

- Close the lid and suspend.
- Open the lid. The laptop resumes from suspend as expected. Lights are working.
- Just reboot normally.
- notice after the grub screen, the lights turn off.
- Now pressing FN + space doesn't matter
- To fix, just close the lid, open again, resume and lights are ON

To break the "cycle"

Since i was managing to trigger this everytime, i wonder if a shutdown would break the cycle. And it did.
Steps:

- from previous experiment, close and reopen de lid to get the lights working
- shutdown the laptop
- turn ON again. the lights still work.
- reboot 1 time, the lights still work.
- reboot 1 more time, still works.
- just to make sure, reboot third time. still works.

To enter the cycle again:

- since the previous experiment, instead of rebooting again, just close the lid and let it suspend.
- now open the lid, the lights still work.
- reboot normally, it will break again

My conclusion: suspending/resuming the laptop the first time, triggers something that will happen on the next reboot

tuxedoxt, 2021-08-25:
Hello,

I had a quick go with this procedure and like this it's very reproducible, great job.

I'll have a look later regarding a solution, thanks for the info!

tuxedoxt, 2021-10-01:
Long story short, there were some module init issues preventing code that would already fix the problem to run.

I just pushed the latest restructure (22d450a144a132c9956dfc99d88328f08e28be97), among other things, this should fix the issue, it is not yet packaged but should you test the current repo state feedback is appreciated.

amithiel, 2021-10-02:
> Long story short, there were some module init issues preventing code that would already fix the problem to run.
> 
> I just pushed the latest restructure ([22d450a](https://github.com/tuxedocomputers/tuxedo-keyboard/commit/22d450a144a132c9956dfc99d88328f08e28be97)), among other things, this should fix the issue, it is not yet packaged but should you test the current repo state feedback is appreciated.

Thanks, i'll have to wait for the update to hit AUR :)
cheers

thepasto, 2021-10-02:
> Long story short, there were some module init issues preventing code that would already fix the problem to run.
>
> I just pushed the latest restructure [...]

I just tried master branch, it seems to solve the keyboard backlight issue. Backlight after a couple of reboots, stay on and can be enabled/disabled via keyboard shortcut, as well.

But i found another problem, i don't know if directly related to the module update. TCC doesn't show CPU temp and fans percentage.

[image]

Anyway thanks for your work!

tuxedoxt, 2021-10-04:
> But i found another problem, i don't know if directly related to the module update. TCC doesn't show CPU temp and fans percentage.

Thanks for testing. Check if `modprobe tuxedo_io`  brings back the fan control & temps readout. If installed the module should get loaded automatically on boot though. If it doesn't there might be an issue.

thepasto, 2021-10-04:
> Thanks for testing. Check if `modprobe tuxedo_io` brings back the fan control & temps readout. If installed the module should get loaded automatically on boot though. If it doesn't there might be an issue.

tuxedo_io is loaded, 

> lsmod | grep tuxedo
tuxedo_io              20480  0
tuxedo_keyboard        49152  3 tuxedo_io,uniwill_wmi,clevo_wmi
sparse_keymap          16384  2 asus_wmi,tuxedo_keyboard

But i still have no data on cpu temp and fan control.

tuxedoxt, 2021-10-04:
Think I found it, module reload (tuxedo_io) was necessary. One missing ID check on interface use. Should be fixed with 0747e94f897657bc4529e7601b803f441c3ed108.

thepasto, 2021-10-04:
It's ok now, thanks!

[image]

is there any sysfs documentation for this tuxedo_keyboard module? I didn't find any match in the readme and actual sysfs.
Thanks once again

tuxedoxt, 2021-10-04:
> is there any sysfs documentation for this tuxedo_keyboard module? I didn't find any match in the readme and actual sysfs.
Thanks once again

There's some info regarding configuration of the rgb backlit keyboards. Nothing that applies to the Pulse though.

closed 2021-10-19 by tuxedoxt

amithiel, 2021-10-19:
Working like a charm. thanks so much, i'm finally 100% happy with this laptop :)