tuxedocomputers/tuxedo-keyboard #92: Apply changes in "tuxedo_keyboard.conf" immediately without reboot

roundoutandabout, 2021-06-06:
According to the [article](https://www.tuxedocomputers.com/en/Infos/Help-Support/Instructions/Installation-of-keyboard-drivers-for-TUXEDO-Computers-models-with-RGB-keyboard-.tuxedo) from Tuxedo website, to apply changes in `/etc/modprobe.d/tuxedo_keyboard.conf` without reboot, it is needed to remove and load the module:
`$ sudo rmmod tuxedo_keyboard`
`$ sudo modprobe tuxedo_keyboard`

However the first command gives an error:
`rmmod: ERROR: Module tuxedo_keyboard is in use by: clevo_acpi tuxedo_io clevo_wmi`

I think the right way will be first to remove the modules `clevo_acpi`, `tuxedo_io`, `clevo_wmi` and then `tuxedo_keyboard`.

tuxedoxt, 2021-06-07:
Hello,

yes, you are correct. The article seems to need an update.

For reference here's a link to one install script that does just what you describe.

https://github.com/tuxedocomputers/tuxedo-keyboard/blob/master/deb/module-name/DEBIAN/postinst#L59-L68

roundoutandabout, 2021-06-07:
Thanks,
The issue can be closed.

commit 66bb44c4 (2023-09-21): Merge branch 'auragen3-keys' into 'master'

Aura Gen 3 keys

Closes #92

See merge request tuxedocomputers/development/packages/tuxedo-keyboard!67
closed 2021-06-07 by roundoutandabout