tuxedocomputers/tuxedo-keyboard #98: Is Book XP15 Gen 12 supported by tuxedo_keyboard on debian ?

Armage, 2021-08-11:
Hi,

I have acquired Tuxedo Book XP15 Gen12 and install debian unstable on it. I'm using debian/sid with gnome3.
I've tried to use the tuxedo_keyboard module on it but without success.

Compilation and install (`make dkmsinstall`) seem ok.

```bash
# lsmod | grep tux
tuxedo_io              20480  0
tuxedo_keyboard        49152  3 clevo_acpi,tuxedo_io,clevo_wmi
sparse_keymap          16384  2 intel_hid,tuxedo_keyboard
wmi                    36864  6 tuxedo_io,intel_wmi_thunderbolt,clevo_wmi,wmi_bmof,mxm_wmi,tuxedo_keyboard
```

But when I want to change backlight intensity (Fn +) I only have the "backlight logo" (sort of half-sun) without any bar (which should indicate intensity). And in `/var/log/message`, I get these messages :

```bash
Aug 11 17:25:40 streaker2 kernel: [205149.792817] atkbd serio0: Unknown key pressed (translated set 2, code 0x6f on isa0060/serio0).
Aug 11 17:25:40 streaker2 kernel: [205149.792820] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
Aug 11 17:25:40 streaker2 gsd-power[1649]: gsd_power_backlight_abs_to_percentage: assertion 'max > min' failed
Aug 11 17:25:41 streaker2 kernel: [205149.901072] tuxedo_keyboard: Set brightness on 255
Aug 11 17:25:41 streaker2 kernel: [205150.300580] atkbd serio0: Unknown key released (translated set 2, code 0x6f on isa0060/serio0).
Aug 11 17:25:41 streaker2 kernel: [205150.300584] atkbd serio0: Use 'setkeycodes 6f <keycode>' to make it known.
```

Is there a way to make this working on my setup ?

tuxedoxt, 2021-08-11:
Hello,

`tuxedo-keyboard` is known to work on most kernels. If you mean *keyboard* backlight there is an additional package that is still distributed separately; `tuxedo-keyboard-ite`. Try to install that one as well. It will soon be included in the same package.

Armage, 2021-08-11:
Thx for your answer, @tuxedoxt . And yes, I meant keyboard backlight :) 
Where can I find this tuxedo_keyboard_ite package ? 

By the way, you speak of a `package` ? Is there a tuxedo_keyboard debian `package` I could use ? A repo ?

tuxedoxt, 2021-08-11:
We provide ubuntu repos + mirrors with deb packages.

https://www.tuxedocomputers.com/en/Infos/Help-and-Support/Instructions/Add-TUXEDO-Computers-software-package-sources.tuxedo

Can for example be manually downloaded through http://deb.tuxedocomputers.com/ubuntu/pool/main/t/

tuxedoxt, 2021-08-11:
Great :)

closed 2021-08-11 by tuxedoxt