tuxedocomputers/tuxedo-keyboard #100: Changes in tuxedo_keyboard.conf no longer apply instantly after rmmod/modprobe

eeickmeyer, 2021-08-20:
With regards to Issue #92, I am presently unable to apply the changes to the keyboard on a Clevo P95 using the documented method. Method is as follows:

```
{changes to /etc/modprobe.d/tuxedo_keyboard.conf}
rmmod tuxedo_io
rmmod clevo_wmi
rmmod clevo_acpi
rmmod tuxedo_keyboard
modprobe tuxedo_keyboard 
modprobe clevo_wmi
modprobe clevo_acpi
modprobe tuxedo_io
```

Expected: Changes take effect instantly
Actual: No changes until reboot.

This issue appeared somewhere between the releases of tuxedo-keyboard 3.0.3 and 3.0.5.

Let me know if there's anything I can do to help resolve this issue.

eeickmeyer, 2021-08-23:
I have reason to believe this may have something to do with commit f4622e28f14cbd3602c8c4903fe7edd5934458a2 in which it prevents a race condition. However, this comes at the expense of not reinitializing the device, which is necessary for the changes to tuxedo_keyboard.conf, such as keyboard color and brightness, to take effect.

If there was a command line utility that made this work, then I'd surely use that, but this method I mentioned above is the only way that is documented.

tuxedoxt, 2021-08-24:
Hello,

I reproduced the issue, however it seems like it does try to set (write) the right color but nothing happens. I'll make a note of the issue.

Meanwhile, maybe what you were really looking for is the sysfs interface for making "on the fly" changes. For your device you can find it in `/sys/devices/platform/tuxedo_keyboard`.

Example:
```
echo 0x0000ff | sudo tee /sys/devices/platform/tuxedo_keyboard/color_left
```
to set the left (or whole depending on keyboard) area to blue.

eeickmeyer, 2021-08-24:
Thanks! That actually helped me simplify the script I use to make the changes, as it no longer depends on rmmod/modprobe but still saves the changes to tuxedo_keyboard.conf so that the changes persist between reboots.

As far as I'm concerned, the problem is solved, but I guessing the behavior of the rmmod/modprobe part still needs to be solved?

closed 2023-12-11 by tuxedo-bot