tuxedocomputers/tuxedo-keyboard #102: OpenRGB support?

xeluxee, 2021-08-31:
[OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB) is a project that aims to be a universal multiplatform RGB LEDs controller for many devices, like mouses, keyboards, motherboards...

[Creating](https://gitlab.com/CalcProgrammer1/OpenRGB#contributing) a compatible controller for Clevo/Tuxedo laptops would be advantageous: it would be multiplatform thanks to OpenRGB nature and keyboard settings would be simpler to control for users, since OpenRGB provides a GUI to control keyboard colours, modes and various settings

blueboy-tm, 2021-11-16:
Hello, use this
https://github.com/blueboy-tm/LinuxBacklightManager/

blueboy-tm, 2021-11-16:
> [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB) is a project that aims to be a universal multiplatform RGB LEDs controller for many devices, like mouses, keyboards, motherboards... [...]

blueboy-tm, 2021-11-16:
> [OpenRGB](https://gitlab.com/CalcProgrammer1/OpenRGB) is a project that aims to be a universal multiplatform RGB LEDs controller for many devices, like mouses, keyboards, motherboards... [...]

Hello, use this
https://github.com/blueboy-tm/LinuxBacklightManager/

nils-van-zuijlen, 2021-11-18:
I would like to add that it may be that [the existing driver for Clevo per key keyboards](https://gitlab.com/CalcProgrammer1/OpenRGB/-/wikis/Clevo-Per-Key-Keyboard) works, and that the "only" modification needed is the USB device ID.

der-ali, 2023-01-16:
This would be the best solution, instead of implementing own client.

closed 2023-12-11 by tuxedo-bot