tuxedocomputers/tuxedo-keyboard #103: Cannot configure backlght on Polaris 15 AMD Gen3

chrosey, 2021-09-05:
I tried installing package over various methods
- apt
- make
- make dkms

Changing values in /etc/modprobe.d/tuxedo_keyboard.conf have no effect. 
modprobe ... <parameter> has no effect
echo ... > sudo tee /sys/... no permission

Setting values like RED in color_left produces log entries on reboot because value is not supported. so it seems to work basically. but  I have no change to configure anything.

What can I do?

StevenSeifried, 2021-09-05:
Are you running Ubuntu? You must install [tuxedo-keyboard-ite](http://deb.tuxedocomputers.com/ubuntu/pool/main/t/tuxedo-keyboard-ite/).

chrosey, 2021-09-06:
if that does also matches for tuxedo-os? i'll try after work

chrosey, 2021-09-06:
tuxedo-keyboard-ite was already installed. also reinstalling did not help

tuxedoxt, 2021-09-06:
Hello,

`tuxedo-keyboard-ite` is actually not needed for Polaris.

Try setting the `color` parameter of `tuxedo-keyboard` to `RED` instead.

There's a provisional interface for on the fly changes in ` /sys/devices/platform/tuxedo_keyboard/`.

Here's a link for reference (see option 2 for polaris): https://www.tuxedocomputers.com/en/Infos/Help-Support/Instructions/Installation-of-keyboard-drivers-for-TUXEDO-Computers-models-with-RGB-keyboard-.tuxedo

chrosey, 2021-09-06:
this seems not correct. if i uninstall `tuxedo-keyboard-ite`, keyboard backlight is turned off.

i tried the other steps mentioned in the url, but i don't even have the path `/sys/devices/platform/tuxedo_keyboard/uw_kbd_bl_color/`

tuxedoxt, 2021-09-06:
Ok, let's check the model

```
cat /sys/class/dmi/id/board_name
```

chrosey, 2021-09-06:
sure
```
cat /sys/class/dmi/id/board_name
GMxZGxx
```

tuxedoxt, 2021-09-06:
Ah, this should be a *Stellaris AMD gen 3*, which would explain why it does need `tuxedo-keyboard-ite`.

Although the keyboard backlight for this model theoretically has more fancy features, the driver only has a very basic interface so far. This will be extended in the future.

chrosey, 2021-09-07:
oh you're right...
sorry, i've read to much about polaris on keyboard-backlighting. 

so there is no misconfiguration but it just does not work yet?

tuxedoxt, 2021-09-07:
Correct, driver for this keyboard so far supports the basics, white "color" with brightness control.

DeRaafMedia, 2022-03-28:
I'm confused I have a Polaris 15 Gen3, This is my keyboard model

```
cat /sys/class/dmi/id/board_name
GMxZGxx
```

After reading this thread I believe I can NOT change the backlight colors (made a few attempts which failed)? Any time line when this will be possible. Would love to change the color in certain situations.