tuxedocomputers/tuxedo-keyboard #104: Simpler animations for simpler backilghts (for example, for Aura 15)

TiGR, 2021-09-07:
As I understand, Aura 15 Gen 1 has simple backlight control which allows only to changing color (`color_left` parameter) and brightness for entire keyboard. So, although you can't make animations across the keyboard (such as wave), but you still can make some simpler animations, for example, when the entire keyboard pulses in single color, or when it smoothly changes colors.

Could we have something like this? Or even better, could we have support for OpenRGB as it is suggested in #102 ?

TiGR, 2021-09-07:
Or, does it actually able to make complex animations, but it is not supported by the driver as it is [in situation with Stellaris](https://github.com/tuxedocomputers/tuxedo-keyboard/issues/103#issuecomment-913858591)?

tuxedoxt, 2021-09-07:
Hello,

off hand I do not know if this one supports the animations. You could try to set the `mode` parameter/sysfs entry to other numbers than 0.

TiGR, 2021-09-07:
Nothing happens when I do that.

blueboy-tm, 2021-11-16:
Hello, use this
https://github.com/blueboy-tm/LinuxBacklightManager/

cybik, 2021-12-15:
> Hello, use this https://github.com/blueboy-tm/LinuxBacklightManager/

Why.

closed 2023-12-11 by tuxedo-bot