tuxedocomputers/tuxedo-keyboard #106: TUXEDO Polaris 15 - Gen3 Not Working

ahmedselim2017, 2021-09-12:
Hi, I bought a rebranded TUXEDO Polaris 15 - Gen3 and installed Arch Linux.  I installed this driver using ```aur/tuxedo-keyboard``` .

I can change the brightness of the keyboard from keyboard shortcuts but when I tried to change it using this tool it doesn't work. I tried to change color using ```sudo modprobe tuxedo_keyboard color 1``` and ```sudo modprobe tuxedo_keyboard color_string RED``` but both of them didn't work.

 Then I added ```options tuxedo-keyboard brightness=25 color_string=RED``` to ```/etc/modprobe.d/tuxedo_keyboard.con``` but it didn't work either.

ahmedselim2017, 2021-09-12:
After adding my model name to ```uniwill_keyboard.h``` as said here (https://github.com/tuxedocomputers/tuxedo-keyboard/issues/88#issuecomment-841077932) the problem was solved.

closed 2021-09-12 by ahmedselim2017

brunoais, 2021-09-19:
@ahmedselim2017 I wouldn't close this issue. Even though you solved for you, the real problem still exists.

ahmedselim2017, 2021-09-19:
But the issue happened because I used a rebranded tuxedo computer and the company who sold it to me changed the model name to the name which they sell. So for a normal tuxedo computer which bought from tuxedo shouldn't have this problem.

brunoais, 2021-09-19:
Good point. I forgot that when I was typing my answer.
Btw, I really doubt it it's a rebranded tuxedo laptop. It's probably a rebranded TongFang laptop. For all I know, Tuxedo doesn't sell rebrands. TongFang sells rebrands, Tuxedo rebrands TongFang laptops into their own.
For example, the Tuxedo Polaris 15 Gen1 4800H+RTX2060 is a TongFang GK5NR0O. But that's just the hardware parts. The firmware is different. I don't know by how much, though.

tuxedoxt, 2021-09-20:
Sounds more or less correct. Firmware adjustments and fixes vary by model. The "TUXEDO versions" are set apart by other things as well though like Linux support, keyboard layouts, visual configurability etc.