tuxedocomputers/tuxedo-keyboard #107: Tuxedo Book XP17 Gen12 don't work

leuchtrakete666, 2021-09-16:
its not working with my machine. 

i read that the driver is checking the main board. 

when using dmidecode it gives me this:

Handle 0x0001, DMI type 1, 27 bytes   
System Information
        Manufacturer: TUXEDO
        Product Name: TUXEDO Book XP15 / XP17 Gen12
        Version: Not Applicable
        Serial Number: Not Applicable
        UUID: bd8444c8-f770-41f2-8c99-a59bc92b5ebe
        Wake-up Type: Power Switch
        SKU Number: XP15_XP1712
        Family: Not Applicable

tuxedoxt, 2021-09-17:
Hello,

are you talking about keyboard backlight? If so, there are many versions and until we have a proper GUI for it you need to use the method that works for your version. Here's a link with some info. *Option 3* should be for your device.

https://www.tuxedocomputers.com/en/Infos/Help-Support/Instructions/Installation-of-keyboard-drivers-for-TUXEDO-Computers-models-with-RGB-keyboard-.tuxedo

It should also be possible to use these parameters *on the fly* through `/sys/module/tuxedo_keyboard_ite/parameters`

leuchtrakete666, 2021-09-17:
yes, i know this already, thank you.

but how to use these parameters? 

how to set individual colors for the keys? like this white with the red t u x e d o keys? what modes are there to be set?

when i change the tuxedo_keyboard.conf and do a reboot, the only thing that work is to set the color for all keys. when i add mode=3, for example, on reboot the keys are set to a very dimmed red. thats by the way always the case, when something is set the driver cannot interpret.  

please give some assistance. thank you

tuxedoxt, 2021-09-17:
Regarding per-key control
- As you have noticed it is prepared for it in the driver but there is no user interface yet.
- This is planned for, incl. TCC integration

Regarding brightness, you're right, the brightness parameter seem to be missing, thanks, I'll make a note.

Otherwise usage of the parameters as they currently are corresponds to setting the color values (0-255) and then writing 1 to `set_color` to apply. This should write the chosen color to all keys.

leuchtrakete666, 2021-09-17:
i took a closer look at the tuxedo_keyboard_ite.c

it seems the t u x e d o in red is hardcoded in the file. what i understood in this file is defined how the fn keys behave. too sad, that coding is not my cup of tea.

but anyway, i am looking forward for a gui to make all the beautiful things happen, what can be done with this keyboard.

unfortunately its not supported by the openrgb project on gitlab.  what is really strange, cause this xp17 or actually clevo PC70DS-D barebone is a very popular chassis that i found offered by many companies. 

i hope you will also take care for the fingerprint sensor some time. it's not supported by the libfprint. but thats another story.

leuchtrakete666, 2021-09-19:
i have no idea of coding, but reading and interpreting code is not so difficult. with some try and error i managed to configure the driver to have 3 more backlight modes. wave, breathe, "i don't know how to name it" and the default random key lights.

furthermore i set the backlight for the keys of my most played game. no man's sky.

in the very beginning of tuxedo_keyboard_ite.c i had to change the count of extra modes. 2 was default, i added 3 modes:

```C
// Amount of extra modes in addition to the color ones
static const int MODE_EXTRAS_LENGTH = 5;
```

thats the section which defines the extra modes of the backlight driver in tuxedo_keyboard_ite.c

```C
static void send_mode(struct hid_device *dev, int mode)
{
	int row, col;
	u8 color_red, color_green, color_blue;
	
	if (dev == NULL) {
		return;
	}

	p_set_color = 0;

	if (mode < MODE_MAP_LENGTH) {
		// Color modes, mode is index to mode_to_color array map
		color_red = (mode_to_color[mode] >> 0x10) & 0xff;
		color_green = (mode_to_color[mode] >> 0x08) & 0xff;
		color_blue = (mode_to_color[mode] >> 0x00) & 0xff;
		keyb_set_all(dev, color_red, color_green, color_blue);
	} else if (mode == MODE_MAP_LENGTH) {
		// No Man's Sky Keys
		for (col = 0; col < KEYBOARD_COLUMNS; ++col) {
			for (row = 0; row < KEYBOARD_ROWS; ++row) {
				if (
					(row == 2 && col == 3) ||   // W
					(row == 3 && col == 2) ||   // A
					(row == 3 && col == 3) ||   // S
					(row == 3 && col == 4) ||   // D
					(row == 4 && col == 0) ||   // Shift
					(row == 5 && col == 5) ||   // Space
					(row == 5 && col == 9) ||   // Space
					(row == 3 && col == 9)      // K
				) {
					keyb_send_data(dev, 0x01, get_led_id(row, col), 0xff, 0xff, 0x00);
				} else if (
					(row == 1 && col == 1) ||   // 1
					(row == 1 && col == 3) ||   // 3
					(row == 2 && col == 2) ||   // Q
					(row == 2 && col == 4) ||   // E
					(row == 3 && col == 5) ||   // F
					(row == 3 && col == 6) ||   // G
					(row == 3 && col == 7) ||   // H
					(row == 4 && col == 3) ||   // Y
					(row == 4 && col == 4) ||   // X
					(row == 4 && col == 5) ||   // C
					(row == 4 && col == 7)      // B
				) {
					keyb_send_data(dev, 0x01, get_led_id(row, col), 0xff, 0xff, 0xff);
				}else {
					keyb_send_data(dev, 0x01, get_led_id(row, col), 0x00, 0x00, 0x00);
				}
			}
		}
	} else if (mode == MODE_MAP_LENGTH + 1) {
		// Breath color animating effect, special mode
		keyb_send_data(dev, 0x00, 0x02, 0x00, 0x00, 0x00);
	} else if (mode == MODE_MAP_LENGTH + 2) {
		// Wave color animating effect, special mode
		keyb_send_data(dev, 0x00, 0x04, 0x00, 0x00, 0x00);
	} else if (mode == MODE_MAP_LENGTH + 3) {
		// Random color animating effect, special mode
		keyb_send_data(dev, 0x00, 0x09, 0x00, 0x00, 0x00);
	} else if (mode == MODE_MAP_LENGTH + 4) {
		// Flashing color animating effect, special mode
		keyb_send_data(dev, 0x00, 0x0a, 0x00, 0x00, 0x00);
	}
}
```
after changing the files i reinstalled the moduls with dkms.

work fine.

but of course it would be nice to have a gui to do this changes on the fly and maybe define backlight profiles for certain apps. unfortunately i cannot code something like this. i was happy to manage what i did with the driver file.

tuxedoxt, 2021-09-20:
Yeah, great that you managed to implement some things that you like. As you found out, even though it is not on github yet the source is distributed in the package. Only reason it didn't get its own github project yet is that it will soon be part of the `tuxedo-keyboard` project as a module collection.

closed 2023-12-11 by tuxedo-bot