tuxedocomputers/tuxedo-keyboard #109: Can't compile for 5.13.* kernel

moobsen, 2021-09-17:
Hi,
when I try to compile the module for my custom 5.13.* kernel I get the following errors:

```
mo@mo-TUXEDO-Aura-15-Gen1:~/src/github/tuxedo-keyboard$ make clean && make
make -C /lib/modules/5.13.17-moobsen/build M=/home/mo/src/github/tuxedo-keyboard clean
make[1]: Entering directory '/usr/src/linux-headers-5.13.17-moobsen'
  CLEAN   /home/mo/src/github/tuxedo-keyboard/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.13.17-moobsen'
make -C /lib/modules/5.13.17-moobsen/build M=/home/mo/src/github/tuxedo-keyboard modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.17-moobsen'
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_keyboard.o
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_wmi.o
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_acpi.o
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.o
  MODPOST /home/mo/src/github/tuxedo-keyboard/Module.symvers
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_acpi.mod.o
  LD [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_acpi.ko
  BTF [M] /home/mo/src/github/tuxedo-keyboard/./src/clevo_acpi.ko
Skipping BTF generation for /home/mo/src/github/tuxedo-keyboard/./src/clevo_acpi.ko due to unavailability of vmlinux
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_wmi.mod.o
  LD [M]  /home/mo/src/github/tuxedo-keyboard/./src/clevo_wmi.ko
  BTF [M] /home/mo/src/github/tuxedo-keyboard/./src/clevo_wmi.ko
Skipping BTF generation for /home/mo/src/github/tuxedo-keyboard/./src/clevo_wmi.ko due to unavailability of vmlinux
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.mod.o
  LD [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko
  BTF [M] /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko
Skipping BTF generation for /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_io/tuxedo_io.ko due to unavailability of vmlinux
  CC [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_keyboard.mod.o
  LD [M]  /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_keyboard.ko
  BTF [M] /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_keyboard.ko
Skipping BTF generation for /home/mo/src/github/tuxedo-keyboard/./src/tuxedo_keyboard.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.13.17-moobsen'
```
I'm not sure what it means that vmlinux is not available.

tuxedoxt, 2021-09-17:
Hello, sorry I do not know what it could be. Maybe check your config for BTF settings or out of tree module settings that might stand out. Maybe compare to a similar ubuntu kernel config (which is what we are normally using and testing against).

Are the modules still built correctly?

moobsen, 2021-09-17:
It's most likely due to some breaking kernel config changes on my part. (I activated code signing without really knowing what I was doing) I'm not sure if the modules were built correctly, but I am assuming yes.

Am going to close this as the module build with DKMS works just fine. Sorry for the noise.

"Fix" for future reference:
`make clean && sudo make dkmsinstall`