tuxedocomputers/tuxedo-keyboard #110: [Request] Setting a timeout for the keyboard.

onurbbr, 2021-09-25:
I'm using a Tuxedo Polaris 17 Gen1 clone. Is there a possibility to set the timeout on the keyboard? When we do not do anything, the keyboard should turn off its light, but the keyboard does not turn off at all. Will there be work on this in the future?

thom-10, 2021-11-05:
This is something I'm missing very much, too (InfinityBook Pro 15 here)

XanderThunder, 2021-11-06:
I‘d really appreciate such a feature, aswell! That this wasn‘t a thing with Tuxedo-Laptops kind of surprised me because all the other stuff is so awesome!

tuxedoxt, 2021-11-15:
Hello,

this is one of the more requested features. Keyboard backlight is due for improvements and TCC integration soon.

As for the driver part, the idea is to finally expose backlight brightness control through a proper `/sys/class/leds` `::kbd_backlight` interface. (Already exist for some models but the majority still needs.) This alone would allow userspace to do the timeout.

Not all distributions have these features, however, and there could be need for a timeout option in the driver **or** TCC as well. 

Any thoughts?

cstamas, 2021-11-15:
IMHO, exposing it first via sysfs is completely okay (and later see is some userpace solution okay or more needed, like touchpad userpace fix)

blueboy-tm, 2021-11-16:
Hello, use this
https://github.com/blueboy-tm/LinuxBacklightManager/

Uhter, 2021-11-17:
> Hello, use this
> https://github.com/blueboy-tm/LinuxBacklightManager/

Hello @blueboy-tm, I've installed it but how to launch it?

blueboy-tm, 2021-11-17:
hi
it is activated automatically after installation
If this has not happened
run app on opt
/opt/lbm-1.0.0-beta/
./LinuxBacklightManager

‫‪l·inadapté‬‏ ***@***.***‬‏> در تاریخ چهارشنبه ۱۷ نوامبر
۲۰۲۱ ساعت ۱۳:۰۲ نوشت:‬

Uhter, 2021-11-17:
After research, LinuxBacklightManager is nowhere.
Maybe an install issue (from me). I don't understand either how to uninstall.

onurbbr, 2022-07-20:
@tuxedoxt I created my timeout project for tuxedo and clevo models with one-way keyboard backlight. Seems to be ok now.
This is the [link](https://github.com/onurbbr/tuxedo-keyboard-tools)

closed 2022-07-20 by onurbbr