tuxedocomputers/tuxedo-keyboard #113: Pulse 15: weird messages in `dmesg` when pressing any special function key (Fn+...)

ileonte, 2021-11-07:
I see messages like these whenever I press any `Fn+...` combination:
```
[   65.554472] atkbd serio0: Unknown key pressed (translated set 2, code 0xf8 on isa0060/serio0).
[   65.554492] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   67.781176] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   67.781196] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   75.941180] atkbd serio0: Unknown key pressed (translated set 2, code 0xf8 on isa0060/serio0).
[   75.941199] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   77.010279] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   77.010298] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   78.210066] atkbd serio0: Unknown key pressed (translated set 2, code 0xf8 on isa0060/serio0).
[   78.210086] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[   79.419898] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[   79.419918] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[  345.024863] atkbd serio0: Unknown key pressed (translated set 2, code 0xf8 on isa0060/serio0).
[  345.024883] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
[  351.004821] atkbd serio0: Unknown key released (translated set 2, code 0xf8 on isa0060/serio0).
[  351.004842] atkbd serio0: Use 'setkeycodes e078 <keycode>' to make it known.
```

As far as I can tell these are mostly useless as everything I've tested works as expected - volume control, screen brightness, keyboard backlight, airplane mode, trackpad on/of - they all work fine but they still print the warnings (?) in `dmesg`.

I'm running kernel `5.15.1` on Gentoo with `tuxedo-keyboard` version `3.0.8` and `tuxedo-cc-wmi` version `0.1.4-r1`. I have seen these messages on older kernels as well (`5.14` for sure) with the same versions of the Tuxedo packages.

liketechnik, 2021-11-15:
The same is the case for me on nixos with kernel `5.14.16` with `tuxed-keyboard` version `3.0.8`.

KallePM, 2021-11-15:
Same problem here.
tuxedo-keyboard: `3.0.9`
kernel: `5.14.16`

The scancode for `Fn` (e078) is not mapped to any keycode.
"Fixed" by assigning keycode 112 (any unassigned keycode will be fine).
Dmesg is no longer flooded and I got another key to map stuff to..

tuxedoxt, 2021-11-15:
Hello,

correct, this event is currently not used directly so you could map it to what you want.

ileonte, 2021-11-15:
I have to admit I would never have expected the `Fn` key to be usable for anything else other than generating 'special' key presses in combination with other keys (I don't think I've seen any other laptop with this behavior).

What can I map it to so that it doesn't interfere with anything else?

KallePM, 2021-11-15:
You can check for unmapped keycodes with `xmodmap -pke`.
Then your can map it with `setkeycodes e078 <keycode>`.
\<keycode> should be the keycode from xmodmap minus 8.

For me keycode 120 was unmapped so I bound Fn to it: `sudo setkeycodes e078 112`

Technologicat, 2023-03-02:
Happens also on a CLEVO PD70PNN1 (`PD5x_7xPNP1_PNR1_PNN1_PNT1`), although not Tuxedo-branded.

On this machine, pressing the `Fn` key generates `translated set 2, code 0x6f on isa0060/serio0`.

So perhaps a feature of CLEVO-based laptops in general?